OPCFoundation/UA-.NETStandard #657: How send file from opc ua client to opc ua server?

TarmaRoving34, 2019-02-19:
Good morning,
I'm creating a C# application that sends a file to a Siemens numeric control, using OPC UA. 
I'm using the CopyFileToServer method. The file is created but I have seen that to pass the raw data in the file, 
you must use the Open method contained in filetype, pass the raw data and call the Close method to close the file. 
I tried several times to use the Open method without having succeeded. Can someone help me?
I tried it in Windows 10 64 bit computer with Visual Studio 2017.
public void SendFile(Opc.Ua.Client.Session session)
        {
            try
            {
                if (session != null)
                {
 
                    NodeId node0 = new NodeId("ns=2;s=/Methods");
                    NodeId node1 = new NodeId("ns=2;s=/Methods/GiveUserAccess");
                    object[] argument0 = new object[2];
 
                    argument0[0] = "MARES";
                    argument0[1] = "SinuWriteAll";
 
                    session.Call(node0, node1, argument0);
 
                    NodeId node = new NodeId("ns=2;s=/Methods");
                    NodeId method = new NodeId("ns=2;s=/Methods/CopyFileToServer");
                    object[] argument = new object[3];
                    byte[] data = new byte[1];
 
                    argument[0] = "Sinumerik/FileSystem/Part Program/sendFile.mpf";
                    argument[1] = data;
                    argument[2] = true;
 
                    var a = session.Call(node, method, argument);
 
                    NodeId nodeFile = new NodeId("file to open"); // The problem is this (i don't find the method for the file server nodeid)
                    NodeId methodOpen = new NodeId("ns=0;i=11580");
                    object[] argument1 = new object[1];
                    argument1[0] = OpenFileMode.Write;
                    
                    var hndl = session.Call(nodeFile, methodOpen, argument1);
                }
            }
            catch (Exception ex)
            {
 
            }
        } 
The code above returns the following exception: 
"Bad invalid argument".
I used .Net Framework 4.7.1, DLL OPC UA (OPC UA Foundation) Ver 1.3.352.0 
 
 
Thanks,
best regards,

AeonSake, 2019-02-19:
A `MethodNode` requires its parent `ObjectNode` in order to call the method.
So your `nodeFile` should have the node ID of the parent node (which must be an ObjectNode).

To get the parent node of a node, you can use something like this:
```csharp
private ReferenceDescription GetMethodParent(Session session, NodeId methodNodeId)
{
  session.Browse(
    null,
    null,
    methodNodeId,
    0u,
    BrowseDirection.Inverse, //browse in inverse direction
    ReferenceTypeIds.HierarchicalReferences,
    true,
    (uint) NodeClass.Object, //only return nodes of type Object
    out _,
    out var references); //get all parent references

  var parentReference = references[0]; //a MethodNode can only have one parent reference
  return ExpandedNodeId.ToNodeId(parentReference.NodeId, session.NamespaceUris); //return the node ID converted from an ExpandedNodeId
}
```

TarmaRoving34, 2019-02-19:
Thanks for reply.
I try this:
```
 NodeId methodOpen = new NodeId("ns=0;i=11580"); // For the Open Method
 var b = GetMethodParent(session, methodOpen);
```
But there is an exception because references count is 0 (error index out of arry/list).

AlinMoldovean, 2019-02-19:
@TarmaRoving34 ,

The problem can that the parent node of the method is not of NodeClass.Object as in your browse filters.
It is NodeClass.ObjectType, in case you call the method on the type tree.
So you have to put:
_(uint) NodeClass.Object, //only return nodes of type Object..._

Or maybe the Open method node does not have the reverse reference to its parent.

But you can get the parent node anyway.
If you browse the server from an OPC UA Client, you should be able to get the parent node of the "Open" method in the context of the FileType object instance you want to use.

Something like this 

[image]

TarmaRoving34, 2019-02-19:
Thanks for reply.
I try this but the error is the same:

 ```
NodeId nodeFile = new NodeId("ns=0;i=86"); //Types Id
                    NodeId open = new NodeId("ns=0;i=11580"); //Open Id
                    object[] argument1 = new object[1];
                    argument1[0] = OpenFileMode.Write;

                    var hndl = session.Call(nodeFile, open, argument1);
```
Is it possible that the first NodeId is the NodeId of the file sent to the Server?

TarmaRoving34, 2019-02-19:
Or is there a different way to get files to the server?

MD-V, 2019-02-20:
Hello @TarmaRoving34,

I have looked at hte documentation [SINUMERIK Documentation](https://cache.industry.siemens.com/dl/files/507/109757507/att_950770/v1/840Dsl_OPCUA_commiss_man_0418_en-US.pdf)
unfortunately the most important information is missing.
Can you please provide the following information so I can provide you an example:

NodeId of the Methods node.
NodeId of the CopyFileToServer node.
NodeId of the Open node in the CopyFileToServer node
NodeId of the Write node in the CopyFileToServer node
NodeId of the Close node in the CopyFileToServer node

TarmaRoving34, 2019-02-20:
Find a solution.

```
 NodeId open = new NodeId("ns=2;s=Sinumerik/FileSystem/Part Program/SENDFILE.MPF.Open");
                    var parent = GetMethodParent(session, open);
                    object[] argument1 = new object[1];
                    argument1[0] = Convert.ToByte(OpenFileMode.Write);
                    var hndl = session.Call(parent, open, argument1);
```
In this way I have the handle to use with the write function.

closed 2019-02-20 by TarmaRoving34

Govinda010, 2022-11-16:
@TarmaRoving34 
Can you share your code? I am also trying to copy the file from client side to the opcua server in siemens but i didn't no find the way to browse and select the file from client side.

MD-V, 2022-11-16:
Hello @Govinda010,
can you elaborate on your problem a little more? I don't think @TarmaRoving34 has browsed the server in his client.