OPCFoundation/UA-.NETStandard #658: Call ReadTag method error (ScanData)

DavidSJakab, 2019-02-20:
Hello!

In C # I write a program for inviting an RfID method. I can enter the input parameters to invite the method, but I cannot specify the (ReadTag method) Identifier / ScanData value shown in the attached image because it is not known for the object. The question would be to use the open source code of the UA-.NET Standard
OPCFoundation / UA .NETStandard
how can I query the Default input value in the image as in the client program?
({0 | System.Byte [] | | {0 | System.Byte [] | 0 | 0} | (null)})
I want a C # example program to call the ReadTag method?

The goal would be to serialize the input and output parameters on a JSON basis. Based on JSON, build the Object [] form of the input parameters and run the method. And so send the outgoing parameters in JSON format.

Thank you in advance for your response.

Best regards:
D.
[image]

MD-V, 2019-02-20:
Hello @DavidSJakab,

it seems that you are using a Server that uses to companion spec of AutoId (Find the NodeSet here:
[Opc.Ua.AutoID.NodeSet2.xml](https://opcfoundation.org/UA/schemas/AutoID/1.0/Opc.Ua.AutoID.NodeSet2.xml))

Unfortunately there is no ModelDesign available for AutoID. As you only need ScanData I created a ModelDesign for that: [ModelDesign](https://pastebin.com/BWax0Yaq)

When running the XML through the ModelCompiler you get the following code: [C#](https://pastebin.com/b0WC56fe)

Copy the code into your application and try the following:

`   EncodeableFactory.GlobalFactory.AddEncodeableType(typeof(ScanData));
var scanDataType = new ScanData{ByteString = null,String = "Hello"};var test = new Variant(scanDataType, TypeInfo.Construct(scanDataType));`

use varaible test as the input for your method.

Dont know if everything works like this. I dont have a server to test.

DavidSJakab, 2019-02-21:
Thank you very much! :)

Best regards,
D.

Michi <notifications@github.com> ezt írta (időpont: 2019. febr. 20., Sze,
17:58):

MD-V, 2019-02-21:
Hello @DavidSJakab,
does it work? 
If yes please close the issue :)

DavidSJakab, 2019-02-21:
Yes, it works! Thank you very much! :)

Best regards,
David

DavidSJakab, 2019-02-24:
Hello!

Does not work on the sharp server. :( Can I request a specific example program that invokes the ReadTag and ScanStart methods?

Thank you very much!

MD-V, 2019-02-25:
Hello @DavidSJakab ,

as I dont have a test server I cannot provide you an example.
If you can please upload your code (client/server) to github so we can have a look.

DavidSJakab, 2019-02-25:
Dear Michi!

Thank you for your help. I can send you a TeamViewer ID for the test server.
What kind of non-public place can I send?

Best regards:
D.
[OPCUACallMethod.zip](https://github.com/OPCFoundation/UA-.NETStandard/files/2899720/OPCUACallMethod.zip)

DavidSJakab, 2019-02-25:
Dear Michi,

When I decompress Server AutoId, I get the attached XML. I manage to connect, read, write, basic method call can succeed. The problem is that I cannot call ScanData and ScanSettings. I tried to create my own object and invite it, but I did not succeed.

[AutoID.txt](https://github.com/OPCFoundation/UA-.NETStandard/files/2899891/AutoID.txt)

MD-V, 2019-02-25:
Hi @DavidSJakab,

please contact me via twitter (Link in my profile) and send my the TeamViewer details.
I will then provide you a repository with example code

DavidSJakab, 2019-02-25:
Thank you for your helps! Perfect work!

[autoidtest.zip](https://github.com/OPCFoundation/UA-.NETStandard/files/2902198/autoidtest.zip)

[image]

closed 2019-02-25 by DavidSJakab

NichtEuler, 2019-03-06:
Hi @DavidSJakab ,
May I ask how you set up your AutoIDServer?
This would be much appreciated.

MD-V, 2019-03-06:
@NichtEuler 
I think @DavidSJakab used the built in OPC UA Server of the SINUMERIK CNC automation system.
So no C# server was created.

DavidSJakab, 2019-03-06:
Szia @NichtEuler  , 

I use the built-in system of the Simatic s7-1500. AutoID is not present by default.

Best regards,
D.

NichtEuler, 2019-03-07:
@MD-V @DavidSJakab 
Thank you for the quick answers.