OPCFoundation/UA-.NETStandard #659: How can I refresh Server Node?

TarmaRoving34, 2019-02-22:
I create a new file on server, but the NodeId of file there isn't.
Is there a method to refresh the Server Node from a client?
Thanks

AlinMoldovean, 2019-02-22:
@TarmaRoving34 ,

Make sure to call AddPredefinedNode() for the node on the server.
On the client side you need a new Browse request()

TarmaRoving34, 2019-02-22:
thanks for the reply,
How can I call a rebrowse from client?
Sorry but I don't have find a documentation for this :-(

TarmaRoving34, 2019-02-22:
I used this method:
 
```
private ReferenceDescriptionCollection Browse(Session session, BrowseDescription nodeToBrowse, bool throwOnError)
        {
            try
            {
                var descriptionCollection = new ReferenceDescriptionCollection();
                var nodesToBrowse = new BrowseDescriptionCollection { nodeToBrowse };
                BrowseResultCollection results;
                DiagnosticInfoCollection diagnosticInfos;
                session.Browse(null, null, 0U, nodesToBrowse, out results, out diagnosticInfos);
                ClientBase.ValidateResponse(results, nodesToBrowse);
                ClientBase.ValidateDiagnosticInfos(diagnosticInfos, nodesToBrowse);
                while (!StatusCode.IsBad(results[0].StatusCode))
                {
                    for (var index = 0; index < results[0].References.Count; ++index)
                        descriptionCollection.Add(results[0].References[index]);
                    if (results[0].References.Count == 0 || results[0].ContinuationPoint == null)
                        return descriptionCollection;
                    var continuationPoints = new ByteStringCollection();
                    continuationPoints.Add(results[0].ContinuationPoint);
                    session.BrowseNext(null, false, continuationPoints, out results, out diagnosticInfos);
                    ClientBase.ValidateResponse(results, continuationPoints);
                    ClientBase.ValidateDiagnosticInfos(diagnosticInfos, continuationPoints);
                }
                throw new ServiceResultException(results[0].StatusCode);
            }
            catch (Exception ex)
            {
                if (throwOnError)
                    throw new ServiceResultException(ex, 2147549184U);
                return null;
            }
        }
```
Is it correct?

fchaxel, 2019-02-24:
Hi,

See #490 

Bye

TarmaRoving34, 2019-03-02:
Thanks for the reply. Now I see the link. ;)

closed 2019-03-08 by TarmaRoving34