OPCFoundation/UA-.NETStandard #661: Failed to start node manager

PrudhviRajC, 2019-02-24:
Hi,
I am using the Unified Automation .NET SDK to implement the OPC UA server. I am trying to import the XML file to my OPC UA Server. Unfortunatey, I am facing with the error below,
[image]

The Server is in good condition but there were nodes missing because of the error. Did anyone has any idea or overcame this error. Please help me.
Thank you very much.

MD-V, 2019-02-26:
@PrudhviRajC you are trying to load a resource in your nodemanager that does not exist. Can you pleas provide sourcecode?

PrudhviRajC, 2019-02-26:
Yes, you are right.. Thank you. I resolved the issue now.

closed 2019-02-26 by PrudhviRajC