OPCFoundation/UA-.NETStandard #662: How to: AutoID

NichtEuler, 2019-02-25:
Hi,

I want to create a very basic AutoID Server. (#658)
So i got myself the DI classes from [here](https://github.com/OPCFoundation/UA-Nodeset/tree/master/DI).
AutoID only provides the [NodeSet2.xml and a NodeID.csv](https://opcfoundation.org/UA/schemas/AutoID/1.0/).
So here is my question: How can I create a server that reads the files and generates a namespace according to those?
The solution to import NodeSet2.xml files i described in #546 doesn't really work here.

MD-V, 2019-02-26:
Good morning. Have a look at https://github.com/OPCFoundation/UA-.NETStandard/issues/658.

You have to reverse engineer the NodeSet2 and convert it to a ModelDesign. And then run the ModelCompiler with your ModelDesign and the csv to generate the C# classes.

NichtEuler, 2019-02-26:
Good morning.
So is there a tool, that can convert NodeSet2 to ModelDesigns?
Or a paid tool that even generates the classes and so on? (UAModeler by Unified Automation?)
Is there a reason, that no ModelDesigns are published?
Why would it be a good idea to reverse engineer when there is much room for mistakes?
Is the NodeSet2 missing any information that are described in the Companion Specification?

MD-V, 2019-02-26:
> So is there a tool, that can convert NodeSet2 to ModelDesigns?
Haven't found one yet.
> Or a paid tool that even generates the classes and so on? (UAModeler by Unified Automation?)
The UAModeler just creates NodeSet files and classes for the UnifiedAutomation Toolkit afaik.
> Is there a reason, that no ModelDesigns are published?
Maybe because AutoID wasn't created with the ModelCompiler so there is no ModelDesign.
> Why would it be a good idea to reverse engineer when there is much room for mistakes?
I haven't said that it is a good idea. It's just the only option that I have found.
> Is the NodeSet2 missing any information that are described in the Companion Specification?
I don't know the spec in detail so I cannot give you an answer.

NichtEuler, 2019-02-26:
Thank you so much @MD-V .

It seems like this SDK is missing some essential parts sadly.

closed 2019-02-26 by NichtEuler

AlinMoldovean, 2019-02-26:
@NichtEuler ,

You do not need to convert NodeSet2 to ModelDesign files.
The NodeSet2 format has all the information needed to handle the custom defined types in your application. See #364 
You can define a generic type that handles the custom data types or generate types by reflection.

NichtEuler, 2019-03-04:
@AlinMoldovean ,
ist it possible to provide a small example?
Thank you

AlinMoldovean, 2019-04-03:
@NichtEuler ,

The implementation of this feature requires some development effort and you can find it in some commercial UA SDKs.
I cannot provide just a small sample code that does the job.