OPCFoundation/UA-.NETStandard #665: Problem to connect to OPCUA server: Could not send an Open Secure Channel

alexisescalard, 2019-02-28:
Hello,

I am struggling with a problem of connection in some machines.
I have a tool to check that the connection to a defined URI works as expected. Here is an example:
opc.tcp://MachineName4862/#Basic256-SignAndEncrypt

According to this URI, the correct endpoint is selected to open a connection to the OPC UA server.
Certificates are stored using the certificate Windows store (X509). Client certificate is generated automatically by the SDK.
It works fine when the tool is started as Administrator in all machines. In some of them, it also works with a standard user (not admin).
But in some machines, with a user which is not administrator, the following errors appear in the logs of OPC UA stack:
**2/25/2019 18:00:39.027 _**EXCEPTION**_ BadTcpInternalError 'Could not send an Open Secure Channel request.' 2/25/2019 18:00:39.027 _**EXCEPTION**_ BadTcpInternalError 'Error establishing a connection: Could not send an Open Secure Channel request.'**

Using SysInternals tools, the following errors appear:
**CreateFile - C:\ProgramData\Microsoft\Crypto\RSA\MachineKeys\82da4e98951a81ca813f8b0c84b6a4fc_f165a72f-47b4-49a5-b200-f352463ae0eb - ACCESS DENIED RegQueryValue - HKLM\SOFTWARE\Microsoft\Cryptography\MachineGuid - BUFFER OVERFLOW**

When the connection is successfully done from another machines, the errors are not present and there is no access to the folder C:\ProgramData\Microsoft\Crypto\RSA\MachineKeys...

The workaround is to delete the certificate in the Windows certificate store, start the tool in admin to let him create a new certificate, then start the tool in a normal way (not admin) with the same user.

Why does the SDK try to create a file in this folder ? What is the condition to create a file in this folder ?
Do you have an idea to solve this issue ?
Thanks for your help

alexisescalard, 2019-03-04:
Any ideas ?

AeonSake, 2019-03-05:
> I'm not sure what causes the error but to answer what he folder is used for: it stores RSA key pairs (public and private key) which are created during asymmetrical encryption. A RSA key pair is a unique set of values with special mathematical properties. You can read more about [here](https://en.wikipedia.org/wiki/RSA_(cryptosystem)). The public key is known to all transmission participants while the private key remains private (who would have thought?). This is done on both sides, Client and Server. A typical asymmetrical encryption procedure looks like this:
>
> 1. Sender encrypts the message with the **public key** of the receiver
> 2. Transmission of the encrypted message
> 3. Receiver decrypts the message with its own **private key**
>
> The above ensures secure transmission of data, also called "Confidentiality". To prevent man-in-the-middle attacks, messages can be signed. This is done with certificates which are encrypted with the **private key** of the sender so the receiver can decrypt it with the **public key** of the sender to ensure that the sender is who they claim to be, which is also called "Authentication".

**TL;DR**
The folder contains RSA key pairs which are generated by the OS when asked for a new RSA key. This is done every time a new transmission is started (or sometimes even for every message).

From your error description my guess is that the OS can't handle that many RSA key generation requests as the OPC UA client tries to open a new transmission repeatedly (buffer overflow).

How are you handling failed connection attempts?
Maybe try adding a timeout after a failed connection attempt.

alexisescalard, 2019-04-30:
Hello, 

Here is a (late) update of this ticket.
We still face this issue. To help you to understand it, I created a sample which connects and disconnects from OPC UA server.
You can find binaries and code here: https://transfernow.net/833z7692ogjv

Here are steps we did in two different machines:

- As administrator, try to connect to an OPC UA server. It will create the application certificate in the dedicated X509 store.
- Close the application
- Start the application with user **non admin**. 

In one machine, the connection will pass and in another it will fail with following error:
"Error establishing a connection: Could not open an Open Secure Channel"
The error seems to appear after a while (some days).

Hope this helps

alexisescalard, 2019-05-09:
Good news !

We found a workaround to fix this issue.

Actually the problem appears when application is started with a standard user (non-admin) or without elevation of privileges for an admin user.

We modified the permissions of the folder **C:\ProgramData\Microsoft\Crypto\RSA\MachineKeys** for the user which starts the application. We added "Read an execute" permission for it. The difference with "Everyone" group is the addition of the permission **"Traverse folder / execute file"**.

Then we started again the application with a standard user and it works ! It is the same when starting the application without elevation of privileges for an admin user.

closed 2019-05-09 by AlinMoldovean