OPCFoundation/UA-.NETStandard #667: how to adopt a different Identity and associated Role

sixdiamants, 2019-03-03:
I run the NetCoreConsoleServer allowing anonymous certificate-based access. I created custom NodeManagers and expose my own address spaces.  
This is just fine for my use case in which I run a limited number of servers.   
However, it seems that any client can modify node values. I wish to restrict access to selected users, preferably by associating a Role with a client application certificate according to OPC-UA Figure 20:
  
[image]  
I understand that the NetCoreConsoleServer doesn't provide full OPC-UA functionality, yet I would like to enhance the server such that I can authorise selected applications to adopt well-known roles (OPC-UA Part 3, Table 2) 
* **Observer** to selected remote client application certificates. 
 * **Engineer** to a local client application that updates nodes  
* **Supervisor** to a client such as UAexpert lest I can call the methods like those exposed in the `Server.ServerConfiguration` node.

**How can this be done ?** Note that I prefer to stick to .NET Core because I run Linux. 

I looked hard at below code but `SystemConfigurationIdentity user = context.UserIdentity as SystemConfigurationIdentity;`  in ConfigurationNodeManager.HasApplicationSecureAdminAccess 
returns null such that I get the exception `BadUserAccessDenied`. Is it possible to elevate a selected certificate to another identity ?

```
        private void HasApplicationSecureAdminAccess(ISystemContext context)
        {
            OperationContext operationContext = (context as SystemContext)?.OperationContext as OperationContext;
            if (operationContext != null)
            {
                if (operationContext.ChannelContext?.EndpointDescription?.SecurityMode != MessageSecurityMode.SignAndEncrypt)
                {
                    throw new ServiceResultException(StatusCodes.BadUserAccessDenied, "Secure Application Administrator access required.");
                }

                // allow access to system configuration only through special identity
                SystemConfigurationIdentity user = context.UserIdentity as SystemConfigurationIdentity;
                if (user == null || user.TokenType == UserTokenType.Anonymous)
                {
                    throw new ServiceResultException(StatusCodes.BadUserAccessDenied, "System Configuration Administrator access required.");
                }

            }
        }
```

fchaxel, 2019-03-04:
Hi,

You can find a starting example in SampleApplications\Workshop\UserAuthentification\Sever.

In OnServerStarted ovverided method (from StandardServer class) add a listener

server.SessionManager.ImpersonateUser += new ImpersonateEventHandler(OnSessionLogin);

and check the identifity (login/passwd or certificate) in OnSessionLogin (accept or reject it).

And for instance when a Write (or Read) is done on a node check context.UserIdentity to allows or not the operation.

Good luck.

sixdiamants, 2019-03-04:
Thanks for the tip; I will investigate this approach and return my experience so others may benefit.

sixdiamants, 2019-03-04:
Your suggestion works. 

In userAuthenticationServer.cs I modified OnServerStarted:

```
        protected override void OnServerStarted(IServerInternal server)
        {
            base.OnServerStarted(server);

            // request notifications when the user identity is changed. all valid users are accepted by default.
            server.SessionManager.ImpersonateUser += new ImpersonateEventHandler(SessionManager_ImpersonateUser);

            server.SessionManager.ImpersonateUser += new ImpersonateEventHandler(SessionManager_onSessionLogin);
        }
```
This is handled as below. For the time being I merely let a session with a given thumbprint assume the `SystemConfigurationIdentity`

```
        private void SessionManager_onSessionLogin(Session session, ImpersonateEventArgs args)
        {
            // Liste de polytechniciens admissibles aux grandes tâches.
            String adminClientCertificateThumbprint = "12E713648445...3187C59DBBA1";

            X509Certificate2 clientCertificate = session.ClientCertificate;

            if (session.ClientCertificate.GetCertHashString().Equals(adminClientCertificateThumbprint))
            {
                Console.WriteLine("De gebruiker met certificaat {0} krijgt adminrechten.", clientCertificate.Issuer);
                UserNameIdentityToken idToken = new UserNameIdentityToken();
                idToken.UserName = "sixdiamants";
                // UserNameIdentityToken userNameToken = args.NewIdentity as UserNameIdentityToken;
                args.Identity = new SystemConfigurationIdentity(new UserIdentity(idToken));

            }
            return;
        }
```

sixdiamants, 2019-03-05:
Tried and tested. Insertion of above code in SampleServer.cs works well. It allows me to  
* connect with a client that presents a trusted certificate (selected by thumbprint)
* call methods authorised to effective user identity equal to SystemConfigurationIdentity

```
        private void SessionManager_onSessionLogin(Session session, ImpersonateEventArgs args)
        {
            String adminClientCertificateThumbprint = "12E713648445EA062681F1DB20DA3187C59DBBA1";

            X509Certificate2 clientCertificate = session.ClientCertificate;

            if (session.ClientCertificate.GetCertHashString().Equals(adminClientCertificateThumbprint))
            {
                UserNameIdentityToken idToken = new UserNameIdentityToken();
                idToken.UserName = "trusted UA Expert";
                args.EffectiveIdentity = new SystemConfigurationIdentity(new UserIdentity(idToken));

            }
            return;
        }
```

MD-V, 2019-03-06:
@sixdiamants Please close the issue if your question was answered :)

Thank you

closed 2019-03-06 by sixdiamants