OPCFoundation/UA-.NETStandard #670: FiniteStateMachineState.DoTransition() thread safety problem

Hevgen, 2019-03-06:
A transition should be an atomic operation. At least a few transitions shouldn't be done at the same moment.
But current implementation is not thread-safe.
I propose to use simple or double locking inside FiniteStateMachineState.DoTransition() method. That prevent execution of `ServiceResult status = this.InvokeCallback(this.OnBeforeTransition, context, (StateMachineState) this, transitionId, causeId, inputArguments, outputArguments);` and varibles setting from different threads at the same moment.
Alternatively FiniteStateMachineState.DoTransition() may be done virtual. That allows to use locks only if a programmer really needs it.

Using of locks at the external code (that call DoTransition()) doesn"t cover all situations and less effective from performance side.

EthanChangAED, 2023-08-04:
Hello @Hevgen, are you still seeing this issue?

Hevgen, 2023-08-06:
Hello. 
Yes. The issue is valid. I don't see any changes in source codes related to it.

PR #2988 (2025-02-12): Make FiniteStateMachine DoTransition virtual 
## Proposed changes

Add Lock to finiteStateMachineDoTransition 

## Related Issues

- Fixes #670 

## Types of changes

What types of changes does your code introduce?
_Put an `x` in the boxes that apply. You can also fill these out after creating the PR._

- [ ] Bugfix (non-breaking change which fixes an issue)
- [ ] Enhancement (non-breaking change which adds functionality)
- [ ] Test enhancement (non-breaking change to increase test coverage)
- [ ] Breaking change (fix or feature that would cause existing functionality to not work as expected,  [... cut]
closed 2025-02-12 by mrsuciu