OPCFoundation/UA-.NETStandard #671: How to set Redundancy ServerUriArray

vepes, 2019-03-11:
Dear all,

We have a project where we are updating stack from old codebase. In old code we are setting the redundant server array list in OnServerStarted() callback after server instantiated by searching the "ServerUriArray" child object and then converting it to PropertyState object.

 private void SetServerUriArray(string[] uris)
        {
            ServerObjectState serverObject = (ServerObjectState)ServerInternal.DiagnosticsNodeManager.FindPredefinedNode(
                ObjectIds.Server,
                typeof(ServerObjectState));

            if (serverObject != null)
            {
                BaseInstanceState serverUriArray = serverObject.ServerRedundancy.FindChild(
                    ServerInternal.DefaultSystemContext,
                    new QualifiedName("ServerUriArray"));

                if (serverUriArray != null)
                {
                    PropertyState property = (PropertyState)serverUriArray;
                    property.Value = uris;
                }
            }
        }

The problem is that in new stack the ServerUriArray child object is not convertible to PropertyState<string[]> object. Using the BaseInstanceState returned by serverObject.ServerRedundancy.FindChild we are not able to set value of object. 

We attempted to set child value using serverObject.ServerRedundancy.SetChildValue() function, but this does not replace the actual child value.

We need more instructions how to set the SererUriArray in new code?

Thanks!

MD-V, 2019-03-11:
Have you tried to set property.WrappedValue = uris; ?

AlinMoldovean, 2019-03-11:
It should be something like this:
`  NodeState serverUriArray = ServerInternal.DiagnosticsNodeManager.FindPredefinedNode(VariableIds.Server_ServerRedundancy_ServerUriArray, nul`

But it seems the node is of type _BaseInstanceState_ instead of _PropertyState_
and _BaseInstanceState_  does not have the _Value_ property.

vepes, 2019-03-11:
Yes. The problem is that BaseInstanceState hides the value property of underlying object, so this cannot be used.

Then function
```
serverObject.ServerRedundancy.SetChildValue(ServerInternal.DefaultSystemContext, new QualifiedName("ServerUriArray"), new string[] {
                         "opc.tcp:\\test.server:51211",
                         "opc.tcp:\\test.server:51212",
                         "opc.tcp:\\test.server:51213" }, true);
```
does not replace the actual object. So this does not work either.

Next we have attempted to create the ServerUriArray PropertyState object from scratch. Like this:

 ```
PropertyState<string[]> replacement = new PropertyState<string[]>(state);
                    replacement.NodeId = serverUriArray.NodeId;
                    replacement.BrowseName = serverUriArray.BrowseName;
                    replacement.AccessLevel = AccessLevels.CurrentRead;
                    replacement.Description = new LocalizedText("My test object");
                    replacement.DisplayName = serverUriArray.DisplayName;
                    replacement.Historizing = false;
                    replacement.NumericId = serverUriArray.NumericId;
                    replacement.ReferenceTypeId = serverUriArray.ReferenceTypeId;
                    replacement.SymbolicName = serverUriArray.SymbolicName;
                    replacement.TypeDefinitionId = serverUriArray.TypeDefinitionId;
                    replacement.UserWriteMask = serverUriArray.UserWriteMask;
                    replacement.Value = new string[] {
                         "opc.tcp:\\test.server:51211",
                         "opc.tcp:\\test.server:51212",
                         "opc.tcp:\\test.server:51213" };
                    replacement.WrappedValue = new string[] {
                         "opc.tcp:\\test.server:51211",
                         "opc.tcp:\\test.server:51212",
                         "opc.tcp:\\test.server:51213" };
                    replacement.Timestamp = DateTime.UtcNow;
                    replacement.StatusCode = StatusCodes.Good;
                    replacement.ValueRank = ValueRanks.Scalar;
                    replacement.WriteMask = AttributeWriteMask.None;
                    replacement.UserWriteMask = AttributeWriteMask.None;
                    replacement.Initialized = true;

                    serverObject.ServerRedundancy.ReplaceChild(ServerInternal.DefaultSystemContext, replacement);
```
This seems to replace the ServerUriArray object correctly, but then looking at the ServerUriArray with client does not show correct values / availability

[image]

AlinMoldovean, 2019-03-22:
It seems to be an issue with the _Redundancy ServerUriArray_ node in the standard nodeset file.

AlinMoldovean, 2020-11-04:
Duplicate of #1113 
Reported in Mantis 6215. Should be fixed when a new NodeSet with a fix is published.

closed 2020-11-04 by AlinMoldovean