OPCFoundation/UA-.NETStandard #675: How to write diffferent types of value to a list of nodes ?

CrustyApplesniffer, 2019-03-14:
Hello,

I'm trying to write float, boolean, integer values to a list of nodes using the method below:

```
public void WriteItems(IEnumerable<string> itemNames, IEnumerable<dynamic> itemValues)
{
    try
    {
        
        var valuesToWrite = new WriteValueCollection();
        var names = itemNames.ToArray();
        var values = itemValues.ToArray();
        for(var idx = 0; idx < names.Count(); idx++)
        {
            var itemName = names[idx];
            var value = values[idx];
            
            valuesToWrite.Add(new WriteValue {
                AttributeId = Attributes.Value,                        
                NodeId = new NodeId($"ns=2;s={itemName}"),
                Value = new DataValue {
                    Value = value,                            
                    StatusCode = StatusCodes.Good,
                    ServerTimestamp = DateTime.MinValue,
                    SourceTimestamp = DateTime.MinValue
                }
            });
        }
        
        var responseHeader = _session.Write(null, valuesToWrite, out var results, out var diagnosticInfos);

        ClientBase.ValidateResponse(results, valuesToWrite);
        ClientBase.ValidateDiagnosticInfos(diagnosticInfos, valuesToWrite);

        var exceptions = new List<Exception>();
        for (var i = 0; i < results.Count; i++)
        {
            if (StatusCode.IsBad(results[i]))
            {
                exceptions.Add(ServiceResultException.Create(results[i], i, diagnosticInfos, responseHeader.StringTable));
            }
        }
        if (exceptions.Count > 0)
            throw new AggregateException(exceptions);
    }
    catch (Exception e)
    {
        Log.FatalException("FAILED to write OPC nodes", e);
        throw new Exception("FAILED to write OPC nodes", e);
    }
}
```

I call this method as follow :
```
WriteItems( new List<string> {"Node1", "Node2"},  new List<dynamic> {25.5d, (int)15, true });
```

Every time it populates the exceptions list with  BadTypeMismatch
Where can I find an example of writing to a node from diferent type of data (float, double, int, short, bool) ?

Thank you

AlinMoldovean, 2019-03-22:
@CrustyApplesniffer,

Your code sample is ok.
But when writing the value a node attribute it is important to set a value with the type defined by the server.
Otherwise you will get a  `BadTypeMismatch `error from the internal validation of the request on the server side.

CrustyApplesniffer, 2019-03-23:
So I need to read the node first to get the type defined by the server.
Optimization is to read the node only the first time I want to write its value then I keep in a cache the type defined by the server for this node. 
Then, the next time I want to write the node, I'll ge the type from the cache without having to read the node.

closed 2019-03-28 by AlinMoldovean