OPCFoundation/UA-.NETStandard #677: Server's method with custom type as input parameter

wyepez, 2019-03-17:
Hello,

I'm doing a development where I have to communicate with an OPC-UA server and I'm using a commercial library and it works fine.

Now, I want to create an OPC-UA server that simulates the server side and for this, I'm using the UA-.NETStandard library to achieve it and I have been able to re-create all the address space except for one thing: this server has a method which has a structured custom type as input argument.

As a guide, I used the Reference Server where I could find examples of how to create methods but these methods are very simple in its arguments. They receive string, float, integers, not structured arguments.

I'm able to create custom type as variables in the address space but when I try to do it in methods there is something that eludes me because I'm not able to do it.

The way I'm testing whether I'm doing it right or wrong is by trying to call the methods I create using the UaExpert tool.

In the Reference Server, the arguments that a method receive is given by these lines:

```
addMethod.InputArguments.Value = new Argument[]
{
    new Argument() { Name = "Float value", Description = "Float value",  DataType = DataTypeIds.Float, ValueRank = ValueRanks.Scalar },
    new Argument() { Name = "UInt32 value", Description = "UInt32 value",  DataType = DataTypeIds.UInt32, ValueRank = ValueRanks.Scalar }
};
```

So, when I saw this I said "it's easy, it's just to provide an Argument array of object to InputArgument with DataType as my Custom Type and that's it" but no.

I tried this but it didn't work (BTW, for creating the custom type, I used the Model Compiler tool):

```
MoveOrderMethod.InputArguments.Value = new Argument[]
{
    new Argument() { Name = "Move_Order", Description = "", DataType = new NodeId(Model.DataTypes.WMS_Empfang_Move_Order, 3), ValueRank = ValueRanks.Scalar },
};
```

Can someone help me by shedding some light on what should I do in order to achieve what I want?

MD-V, 2019-03-21:
Hello @wyepez ,

can you see your custom created data type in the Adress Space of your server?

If not you have to pass your namespace uri in the constructor of your NodeManager and load the predefined nodes also in your NodeManager.

AlinMoldovean, 2019-03-22:
@wyepez ,

If the custom data types are defined in the server project with corresponding classes generated by Model Compiler, you should first add them in the known types of `EncodeableFactory` with `Factory.AddEncodeableTypes()`.
You can have a look at the DataTypes Server project in `CreateMasterNodeManager()`

```
protected override MasterNodeManager CreateMasterNodeManager(IServerInternal server, ApplicationConfiguration configuration)
        {
            Utils.Trace("Creating the Node Managers.");

            // add the types defined in the quickstart information model library to the factory.
            server.Factory.AddEncodeableTypes(typeof(Quickstarts.DataTypes.Types.VehicleType).Assembly);
            server.Factory.AddEncodeableTypes(this.GetType().Assembly);

            List<INodeManager> nodeManagers = new List<INodeManager>();

            // create the custom node managers.
            nodeManagers.Add(new DataTypesNodeManager(server, configuration));

            // create master node manager.
            return new MasterNodeManager(server, configuration, null, nodeManagers.ToArray());
        }
```

closed 2020-01-16 by AlinMoldovean