OPCFoundation/UA-.NETStandard #678: CheckApplicationInstanceCertificate exception

fchaxel, 2019-03-17:
Hi,

CheckApplicationInstanceCertificate generates an exception when the certificate is wrong (outdated for instance). Quite strange for a method returning a bool !

If the certificate is wrong how can we try to generates a new one since CreateApplicationInstanceCertificate is a private method ?

In CheckApplicationInstanceCertificate some code shows tests about invalid certificate in order to generate a new one automaticaly but a call to CertificateValidator.Validate( ) is not catched and the exception occur inside it.

Bye

MD-V, 2019-03-18:
Hi @fchaxel,

can't you just call: 
` await _ApplicationInstance.CheckApplicationInstanceCertificate(true, 0);` 

Where true is the value for the silent parameter.

AlinMoldovean, 2019-03-22:
@fchaxel ,

This stack has a lot of places where is throwing exceptions due to some different validation errors.
`CheckApplicationInstanceCertificate()` validates the application instance certificate and throws an exception with a specific information about the validation error.
You have the option to suppress it by setting the `silent `parameter to `true`.

fchaxel, 2019-03-22:
Hi  @AlinMoldovean,  @MD-V  

Sorry but both true or false for the parameter silent gives the same exception.

Sure we can catch it, but we don't have a proper way to re-creates the certificate (other than deleting the file 'manualy' and by calling back the Check method).

MD-V, 2019-03-22:
Hello @fchaxel,

da you have more information on the Exception that is thrown? Like StackTrace and Inner-Exception type

fchaxel, 2019-03-22:
CheckApplicationInstanceCertificate made a call to
   CheckApplicationInstanceCertificate static method without catch
      calling CertificateValidator.Validate without a catch

... and in CertificateValidator.Validate we can see that an invalid certificate throw an exception if the certificat is wrong

MD-V, 2019-03-24:
@fchaxel 

can you also post the exact exception type please?

fchaxel, 2019-03-24:
CertificateValidator.cs line 297.

MD-V, 2019-03-24:
@fchaxel Have you tried setting the CertificateValidation event handler? For an example see [](https://github.com/OPCFoundation/UA-.NETStandard/blob/f1cdabc034ce795e930658d5a021385a7815b826/SampleApplications/Samples/NetCoreConsoleServer/Program.cs) line 230

fchaxel, 2019-03-24:
@MD-V 

CertificateValidation event handler is to validate remote client or server.

When the certificate is wrong it has to be changed. No more no less. 

The original question remain the same : how a new certificate can be re-issued since the CreateApplicationInstanceCertificate is private ? ... and removing the file by seeking in the directory structure seems not to be a clean way.

MD-V, 2019-03-25:
Hello @fchaxel ,

I was able to reproduce your problem.

I think the underlying error is in the following line (798) in the [CertificateValidator](https://github.com/OPCFoundation/UA-.NETStandard/blob/c29ceaafb642ecc2b10db046bbfc11543774fe06/Stack/Opc.Ua.Core/Security/Certificates/CertificateValidator.cs#L798)

The serviceresult (in case of outdated certificate StatusCodes.BadCertificateTimeInvalid) would be catched by the Validate method but line 798 just throws a generic exception:
`                   
throw new ServiceResultException(StatusCodes.BadSecurityChecksFailed);
`

AlinMoldovean, 2019-03-25:
@MD-V ,

You can renew the application instance certificate by deleting the existing one first, and after call `CheckApplicationInstanceCertificate()`
You do not need to search for the certificate on the directory structure for that.
The stack has already implementation for ICertificateStore.Delete() method.
You can find an example in `DeleteApplicationInstanceCertificate()`.
The method is public, but you can reuse the code in your application:

```
        /// <summary>
        /// Deletes an existing application instance certificate.
        /// </summary>
        /// <param name="configuration">The configuration instance that stores the configurable information for a UA application.</param>
        private static async Task DeleteApplicationInstanceCertificate(ApplicationConfiguration configuration)
        {
            Utils.Trace(Utils.TraceMasks.Information, "Deleting application instance certificate.");

            // create a default certificate id none specified.
            CertificateIdentifier id = configuration.SecurityConfiguration.ApplicationCertificate;

            if (id == null)
            {
                return;
            }

            // delete private key.
            X509Certificate2 certificate = await id.Find();

            // delete trusted peer certificate.
            if (configuration.SecurityConfiguration != null && configuration.SecurityConfiguration.TrustedPeerCertificates != null)
            {
                string thumbprint = id.Thumbprint;

                if (certificate != null)
                {
                    thumbprint = certificate.Thumbprint;
                }

                using (ICertificateStore store = configuration.SecurityConfiguration.TrustedPeerCertificates.OpenStore())
                {
                    await store.Delete(thumbprint);
                }
            }

            // delete private key.
            if (certificate != null)
            {
                using (ICertificateStore store = id.OpenStore())
                {
                    await store.Delete(certificate.Thumbprint);
                }
            }
        }
```

MD-V, 2019-03-25:
@AlinMoldovean 

thank you, so the solution should look something like this

               await application.ApplicationConfiguration.Validate(ApplicationType.Server);
                
               bool valid = true;

                try
                {
                    // check the application certificate.
                    await application.CheckApplicationInstanceCertificate(false, 0);
                }
                catch (Exception e)
                {
                    valid = false;
                }

                if (!valid)
                {
                    await DeleteApplicationInstanceCertificate(application.ApplicationConfiguration);

                    application.ApplicationConfiguration.SecurityConfiguration.ApplicationCertificate.Certificate = null;
                }

                // check the application certificate.
                await application.CheckApplicationInstanceCertificate(false, 0);

                var server = new CustomServer();

                // start the server.
                await application.Start(server);

Am I right?

AlinMoldovean, 2019-03-25:
Right, something like that.

closed 2019-03-25 by fchaxel