OPCFoundation/UA-.NETStandard #683: Can't sample faster than publishing rate with TopServer (kepwareEx?)

btrepp, 2019-03-25:
Hi all,

Not sure if this is me mis-understanding the terminology, or an issue with my code or the server.

I've noticed I can't get values at any higher resolution than the subscription 'PublishingInterval', with monitoredItems.

Eg I will set the SamplingInterval to 100 or 500 and the publishing interval to 1000, and I get values every second. I am using the FastDataChangeNotificationEventHandler, and iterating through the MonitoredItemNotification events, calling DequeueValues on the MonitoredItemHandles obtained by 'FindItemByClientHandle'. The queue size is set to 100, so there should be the possibility of multiple values, but dequeuevalues only ever has 1.

Slowing publishing interval to 5000 gives me values every 5 seconds, but still only 1 monitored item event. The data is changing.

Am I misinterpreting these numbers?, and SamplingInterval must always be a minimum of Publishing interval?, if so why is there a queue if the values are always 1?. Or is this possibly something to do with the OPCUA server (TopServer, which I think is rebranded Kepware), and it's sample tags?.

MD-V, 2019-03-25:
Hi @btrepp,

not sure if I'm correct on this one:
When using the FastDataChangeNotificationEventHandler you should receive all elements that are in the queue on one call.  Just iterate through (DataChangeNotification.MonitoredItems). No need to dequeue the notifications manually.

I don't know if queue functionality has to be implemented by the server. Maybe your server doesn't support queuing...

Additional I strongly recommend that you read the following article on [OPC UA Sessions, Subscriptions and Timeouts](https://www.prosysopc.com/blog/opc-ua-sessions-subscriptions-and-timeouts/)

AlinMoldovean, 2019-03-27:
Hi @btrepp ,

You need to make sure the server has support for monitored items queuing.
First step is to check the "revised" values returned by the server, i.e. check monitoredItem.Status.QueueSize after the item was created.

[image]

closed 2019-06-20 by AlinMoldovean

btrepp, 2019-08-20:
So I got a chance to look at this, And I can see via OPCUA that the server is actually sending back  a publish response with multiple samples.

What I don't understand is that the FastDataChangeNotificationEventHandler, responds with
a DataChangeNotification, from here I can get the MonitoredItemsNotificationCollection, but this contains one point (despite the wire having many), so I thought maybe the item notification has multiple points?
The MonitoredItemNotification has only has .Value, for one point, so somewhere the library is confusing me. I either expect multiple MonitoredItemNotifications per "tag" per publish cycle, or a 'Values' on the MonitoredItemNotification. 

Through looking at the source and random experimentation, It appears you need to set CacheQueueSize on the subscription, and then get the MonitoredItem and call DequeueValues() on it.