OPCFoundation/UA-.NETStandard #686: OPC COM Server Wrapper supported profiles

phoenixoflight, 2019-03-27:
Hello there,

I need to know what profiles that the OPC COM Server Wrapper supports.  Is there a full list available somewhere?  I'm not terribly well versed in UA currently.  I'm interested in these in particular:

- Core Server Facet
- Data Access Server Facet
- SecurityPolicy - Basic128Rsa15
- SecurityPolicy - Basic256
- SecurityPolicy - None
- UA-TCP UA-SC UA Binary
- Base Server Behavior Facet

Thanks

phoenixoflight, 2019-04-09:
Two week bump, anyone know?

AlinMoldovean, 2019-04-12:
@phoenixoflight ,

I think you cannot have the list of profiles for this Wrapper because there is no direct correspondence between OPC UA profiles and Classical COM Servers.
Also, please note that this wrapper is just a basic sample for a bridge to Classical OPC Servers

PaulHunkar, 2019-04-12:
I would second the comment Alin.  The COM wrapper is just an example of a wrapper, it is not a commercial offering.  Classic OPC (DCOM) had a fair amount of variance and thus a generic wrapper that would communicate with all Classic DA servers can be some work. Usually wrappers are geared to specific Classic implementations (i.e. each vendor implement Classic usually in a distinct manner) .  The resulting OPC UA model will have variance, since the underlining data will have variances.  The following Facets are about the OPC UA Communication and all wrappers can implement them for the OPC UA side communication: 
- SecurityPolicy - Basic128Rsa15
- SecurityPolicy - Basic256
- SecurityPolicy - None
- UA-TCP UA-SC UA Binary
The other will vary according to what is available in the classic product and or what is added by the wrapper code.

closed 2019-05-23 by AlinMoldovean