OPCFoundation/UA-.NETStandard #687: Exception while loading application configuaraion

NakWarsi, 2019-03-29:
I am using softing "dataFEEDOpcUaNetStandardSdk2.10.0" to develop UA Client(console application).
I am getting an exception it's loading application configuration.
code:-
``ApplicationInstance application = new ApplicationInstance();
            application.ApplicationName = "ConsoleClient";
            application.ApplicationType = ApplicationType.Client;
            application.ConfigSectionName = "ClientTest1COnfiguration"; 
            ApplicationConfiguration config;
            try
            {
                config = application.LoadApplicationConfiguration(false);
            }
            catch (Exception e)
            {
                Console.WriteLine(e);
                throw;
            }``
exception:-
```Message = "Could not load file or assembly 'System.Windows.Forms, Version=4.0.0.0, Culture=neutral, PublicKeyToken=b77a5c561934e089'. The system cannot find the file specified."```

I can't understand why it needs  System.Windows.Forms while my application is a console application.

kaankazaz, 2019-03-29:
Hello Naushad,

Is it caused by ApplicationConfiguration section of your code? Possibly there is another code-block that references System.Windows.Forms by a using statement and you console application is looking for System.Windows.Forms.dll in the bin folder during runtime just because of that.

Is there a chance that you share a simplified version of the code?

NakWarsi, 2019-03-29:
yes, it causes by LoadApplicationConfiguration only.
[
[image]
](url)
as you can see the image I am just loading the configuration file only.

if I put the same code with OPC Foundation library than it's working fine but with softing it's a bit strange.

AlinMoldovean, 2019-03-29:
@NakWarsi ,

The ` LoadApplicationConfiguration(bool silent)` only works if the configuration name is the value set in `ConfigSectionName ` + `".Config.xml"`.

I would suggest to use the` LoadApplicationConfiguration(string filePath, bool silent) `method instead.

If you are using a commercial product (Softing dataFEED OPC UA .NET Standard SDK) please address the specific questions to the technical support of that product.

In case of that specific product, the loading of configuration if performed using a `UaApplication.Create(configFilePath)` method which is not present in this stack

closed 2019-04-01 by NakWarsi