OPCFoundation/UA-.NETStandard #688: Incorrect Node Id: Read and Write fails

asgaralim, 2019-03-30:
Hello Everyone,

Connect the OPC COM wrapper to clearscada. Connect UA Sample client. Browse the tags: It display the tag as below. There some extra character after nodeID (?56625665).
ns=2;s=2:Example Projects.Water and Wastewater.WWTP.Air Blower 1.AWSTEST.CurrentValue?56625665

Now connect third party(UA Gateway) to clearscada. The node id is dispayed as below:
ns=4;s=Example Projects.Water and Wastewater.WWTP.Air Blower 1.AWSTEST.CurrentValue

1. Why COM wrapper display node id differently.
2. How to read and write to Node using OPC COM wrapper

AlinMoldovean, 2019-04-02:
@asgaralim ,

Maybe that specific COM server does not have support for WriteVQT method.

asgaralim, 2019-04-02:
The write Operation failed due to exception  HRESULT: 0xC0040406, which indicates server is not supporting all version of OPC DA. Update the code so that write call will be routed by  Da20WriteItemValues(requests);

public void Write(WriteRequestCollection requests)
        {
            if (m_supportsIOPCItemIO)
            {
                Da30WriteItemValues(requests);
            }
            else
            {
                Da20WriteItemValues(requests);
            }
        }

Still I have extra character padding issue. Investigating the issues

asgaralim, 2019-04-16:
Could you please help me, why tags are padded with extra characters

[image]

AlinMoldovean, 2019-04-17:
@asgaralim ,

You need to debug inside the code to find the answer.

asgaralim, 2019-04-17:
I debugged the code. see below screen shot where code  add these extra  padding.  I don't get these extra padding, when i connect to opc simulator like matrikon and prosys. But when i connect o clearscada, i get these extra padding. I would like to understand in which context these extra padding will come. Please help me. Below code is where extra character are added

[image]

AlinMoldovean, 2019-04-19:
@asgaralim ,

That is because the DA node is a property tag

closed 2019-10-16 by AlinMoldovean