OPCFoundation/UA-.NETStandard #689: Incorrect Export of namespace index

PMarinov1994, 2019-04-01:
When a NodeStateCollection.SaveAsNodeSet2 is called and nodes are written to xml elements, the reference namespace index (i.e. ns=id) is incorrect. In the UaNodeSetHelpers.cs class the function `private string Export(Opc.Ua.ExpandedNodeId source, NamespaceTable namespaceUris, StringTable serverUris)` checks if the source has a valid NamespaceUri index. If it does it calls `private ushort ExportNamespaceUri(string namespaceUri, NamespaceTable namespaceUris)` otherwise it will call `private ushort ExportNamespaceIndex(ushort namespaceIndex, NamespaceTable namespaceUris)`. The two functions do the exact same job, only in different ways. When a new namespace needs to be added in the <NamespaceUris> xml list, a correct namespace index needs to be assigned. At the end the ExportNamespaceIndex function returns the newly created array with namespaces's count as the index, which is correct, but the ExportNamespaceUri function returns the count + 1 (`return (ushort)(count + 1);`). I think that the + 1 should not be there. This result in an incorrect reference to a namespace index that is not present in the <NamespaceUris> list.

PMarinov1994, 2019-04-10:
Also I have notices that when a Opc.Ua.NodeStateCollection with no nodeStates inside (i.e. Items.Count == 0) the exported xml does NOT contain the `<NamespaceUris></NamespaceUris>` xml node that defines the name of this node set namespace uri. The exported xml is useless without the name of namespace of the collection.

cristipogacean, 2021-09-15:
@PMarinov1994, thank you for bringing this up! I can confirm the return (ushort)(count+1) is a bug and will be fixed shortly.
Regarding the second point you brought up -  Opc.Ua.NodeStateCollection with no nodeStates - we would like to get a bit more details on the use-case. If there are no nodes in collection, I believe there are no namespaces to be exported as well, therefore nothing to export.

closed 2021-09-16 by cristipogacean