OPCFoundation/UA-.NETStandard #691: How to use certificate authentication with the server aggregator?

cponcedeleon, 2019-04-02:
I succesfully connected a UaServer to the Aggregation server with anonymous authentication and I'm able to browse the namespace and all. Right now I'm trying to connect to a UaServer using Basic256Sha256 security policy and a X509 certificate. What's the correct way to do this? 

I changed the user token policy in the xml config file of the Aggregation server to 
<ua:SelectedUserTokenPolicy>2</ua:SelectedUserTokenPolicy>
which should be certificate?

However when I connect with any client to the Aggregation server using a valid certificate I see the Aggregator's namespace but cannot browse the UaServer namespace. Does the Aggregation server need to have the certificates before connection to servers?

AlinMoldovean, 2019-04-03:
@cponcedeleon ,

You should check in `AggregationNodeManager.GetClientSession(ServerSystemContext context)` if the connection with the underlying OPC UA Serever is created with success here:

https://github.com/OPCFoundation/UA-.NETStandard/blob/292c7352baf8126f9632c77add678caebb438d60/SampleApplications/Workshop/Aggregation/Server/AggregationNodeManager.cs#L1156

cponcedeleon, 2019-04-04:
It doesn't throw exceptions but `userIdentity` is null, I assume that should contain certificate data. 

`Opc.Ua.Client.Session GetClientSession(ServerSystemContext context)` receives context as null.

AlinMoldovean, 2019-04-12:
@cponcedeleon ,

I think you need to implement the validation of the user identity in your server in the same way as `UserAuthentication `Server project in order to transport the userIdentity to the underlying server.

Something like this:

```
        /// <summary>
        /// Called when a client tries to change its user identity.
        /// </summary>
        private void SessionManager_ImpersonateUser(Session session, ImpersonateEventArgs args)
        {
            // check for a user name token.
            UserNameIdentityToken userNameToken = args.NewIdentity as UserNameIdentityToken;

            if (userNameToken != null)
            {
                VerifyPassword(userNameToken.UserName, userNameToken.DecryptedPassword);
                args.Identity = new UserIdentity(userNameToken);
                Utils.Trace("UserName Token Accepted: {0}", args.Identity.DisplayName);
                return;
            }

            // check for x509 user token.
            X509IdentityToken x509Token = args.NewIdentity as X509IdentityToken;

            if (x509Token != null)
            {
                VerifyCertificate(x509Token.Certificate);
                args.Identity = new UserIdentity(x509Token);
                Utils.Trace("X509 Token Accepted: {0}", args.Identity.DisplayName);
                return;
            }
        }
```

Langer911, 2019-04-22:
Hi Alin

I'm trying the same thing as @cponcedeleon, when I tried to connect with a certificate I received the context null in the GetClientSession method but when I use anonymous authentication, I am receiving the context as serverSystemContext and it connects succesfully. 

[image]

This is the Call Stack when I use anonymous authentication, as you can see it's sending the context different than null and when I try to use certificates it doesn't call all those methods.

[image]

This is the Call Stack with certificates and it is sending the context null.

Do you know what can be the problem here?

Thank you

AlinMoldovean, 2019-04-25:
@Langer911 ,

You need to make a debug inside the stack and find the place where `context `is set.

closed 2019-10-16 by AlinMoldovean