OPCFoundation/UA-.NETStandard #692: HTTPS channel not conformant with Part2 §4.10

ccasteyde, 2019-04-03:
Hello
We seem to have at least one issue with either the Part2 or the stack code.
Part 2 §4.10 basically says that with https, there is no application authentication. However, StandardServer.cs code for CreateHttpsServiceHost requires at least SignEncrypt binding.
This is not conformant, only "None" SecurityMode binding should be exposed if we want to follow the specs. In previous versions, None was automatically added, it was removed without notice but the part 2 and 4 (for registering to LDS) at least were not updated.
Using SignEncrypt ensures we can authenticates the client with HTTPS, but this could also be done by requiring it on the Https channel.
What should be done?

ccasteyde, 2019-04-03:
Please also see #182 on UA Legacy, which has either a vulnerability, or is compliant.

AlinMoldovean, 2019-04-24:
Hi @ccasteyde ,

The specification says in Part2 4.10:

> In OPC UA HTTPS can be used to create Secure Channels, however, these channels do not provide Application Authentication.

But this is for **CreateSecureChannel** step.
An the secure channel is created without application authentication.

The specification at Part6 7.4.1 says:

> Applications which use the HTTPS transport and require application
authentication shall check application Certificates during the CreateSession/ActivateSession
handshake.

And this is what this library does. It validates the client certificate at **CreateSession** step.

Also in Part6 7.4.1 it says:

> HTTPS allows Clients to have certificates; however, they are not required by the HTTPS
transport. A Server shall allow Clients to connect without providing a Certificate during
negotiation of the HTTPS connection.

So this needs to be further reviewed.

ccasteyde, 2019-05-22:
Hello
The specification has indeed addressed the problem.
In Part 6 7.4.1 it also says:
> A SecurityPolicy of None indicates that the Nonces do not need to be signed. The SecurityMode is set to Sign unless the SecurityPolicy is None; in this case the SecurityMode shall be set to None.
Therefore the SignAndEncrypt binding forced by the stack is not the right one. It should depend on the SecurityPolicy asked by the client:
- if 0/None, SecurityMode should be set to None;
- if higher, SecurityMode should be set to "Sign".
The Encrypt part is probably transferred to the TLS channel and should not be used.
If Sign is used, probably the compatibility imposes to use the lowest security mode configured on the server, since it seems only one binding is allowed.
I have code to implement that behaviour and will post it later.

ccasteyde, 2019-07-08:
We choose to let the user tell what SecurityMode it wants, with whatever SecurityPolicy.
He can restrict the combination to the ones that are supported, however if he specifies several ones, we chose highest SecurityMode with weakest SecurityPolicy (because our configuration go from "most compatible (all security policies)" to "most secure one (Basic256Sha256 only)").
Of course the proposed code would depend on how the configuration is proposed, if security policies can be selected by the user independently, it's probably better to choose the most secured one.

Change in Core/Stack/Server/ServerBase.cs, method CreateHttpsServiceHost:

`
        protected List<EndpointDescription> CreateHttpsServiceHost(
            IDictionary<string, ServiceHost> hosts,
            ApplicationConfiguration configuration,
            BindingFactory bindingFactory,
            IList<string> baseAddresses,
            ApplicationDescription serverDescription,
            List<ServerSecurityPolicy> securityPolicies)
        {
            // generate a unique host name.
            string hostName = String.Empty;

            if (hosts.ContainsKey(hostName))
            {
                hostName = "/Https";
            }

            if (hosts.ContainsKey(hostName))
            {
                hostName += Utils.Format("/{0}", hosts.Count);
            }

            // build list of uris.
            List<Uri> uris = new List<Uri>();
            EndpointDescriptionCollection endpoints = new EndpointDescriptionCollection();

            // create the endpoint configuration to use.
            EndpointConfiguration endpointConfiguration = EndpointConfiguration.Create(configuration);
            string computerName = System.Net.Dns.GetHostName();

            for (int ii = 0; ii < baseAddresses.Count; ii++)
            {
                if (!baseAddresses[ii].StartsWith(Utils.UriSchemeHttps, StringComparison.Ordinal) &&
                    !baseAddresses[ii].StartsWith(Utils.UriSchemeNoSecurityHttp, StringComparison.Ordinal))
                {
                    continue;
                }

                UriBuilder uri = new UriBuilder(baseAddresses[ii]);

                if (uri.Scheme == Utils.UriSchemeNoSecurityHttp)
                {
                    uri.Scheme = Utils.UriSchemeHttp;
                }

                if (uri.Path[uri.Path.Length-1] != '/')
                {
                    uri.Path += "/";
                }

                if (String.Compare(uri.Host, "localhost", StringComparison.OrdinalIgnoreCase) == 0)
                {
                    uri.Host = computerName;
                }

                uris.Add(uri.Uri);

                if (uri.Scheme == Utils.UriSchemeHttps)
                {
                    // can only support one policy with HTTPS so pick the best one.
                    ServerSecurityPolicy bestPolicy = null;

                    foreach (ServerSecurityPolicy policy in securityPolicies)
                    {
[... 38 lines cut ...]
                        }
                        if (configExtension != null && "true".Equals(configExtension.InnerText, StringComparison.InvariantCultureIgnoreCase))
                        {
                            for (int i = 0; i < InstanceCertificateChain.Count; i++)
                            {
                                certificateChainList.AddRange(InstanceCertificateChain[i].RawData);
                            }
                        }
                        else
                        {
                            certificateChainList.AddRange(InstanceCertificateChain[0].RawData);
                        }
                        
                        description.ServerCertificate = certificateChainList.ToArray();
                    }
                    
                    description.SecurityMode = bestPolicy.SecurityMode;
                    description.SecurityPolicyUri = bestPolicy.SecurityPolicyUri;
                    description.SecurityLevel = bestPolicy.SecurityLevel;
                    description.UserIdentityTokens = GetUserTokenPolicies(configuration, description);
                    description.TransportProfileUri = Profiles.HttpsBinaryTransport;

                    endpoints.Add(description);
// FOR LEGACY STACK, also remove None Endpoint.
                }`

commit a1e26bf8 (2019-10-02): Interoperability fixes for HTTPS (#814)

* Fixes #709 - Allow HTTPS clients to send the certificate for servers that require TLS client authentication

* Fixes #692 - HTTPS Server listener - Use the first configured policy for HTTPS
closed 2019-10-02 by AlinMoldovean