OPCFoundation/UA-.NETStandard #694: Create NodeSet2.xml file from Address Space

tschombe, 2019-04-04:
Hi,

I have an OPC UA Server that builds its address space from an external system.
I want to save the address space as a nodeset2.xml file to share the address space information offline to a system that can parse the xml file.

I know that model designer generates code by parsing the xml file, but I did not find any code on how to CREATE the xml file.

Can anyone help me with this issue?

AlinMoldovean, 2019-04-04:
@tschombe ,

Check the `NodeStateCollection.SaveAsNodeSet2()` method.
Something like:
```
NodeStateCollection nodesToSave;
// Add the nodes to save

using (Stream stream = File.Open("MyNodeSet2.xml", FileMode.Create, FileAccess.Write, FileShare.Read))
{
    nodes.SaveAsNodeSet2(SystemContext, stream);               
}
```

tschombe, 2019-04-04:
thanks very much for your info,
I will look into this!!!

closed 2019-04-25 by AlinMoldovean

tschombe, 2019-07-12:
after a pause since working on another project i returned to this topic.

I can successfully create a nodeset2.xml File with NodeStatCollection.SaveAsNodeSet2().

But it only contains the node declarations and NOT the corresponding types
( i am using custom types in my address space).

Is there also a way to include the type definitions in the NodeSet?

AlinMoldovean, 2019-07-22:
@tschombe ,

The type definitions can be included in the NodeSet2.xml file by including all the type and type dictionary nodes, in the same way nodeset files from [UA-NodeSet](https://github.com/OPCFoundation/UA-Nodeset) repository.

e.g.

```
 <UAVariable NodeId="ns=1;i=374" BrowseName="1:Opc.MDIS" SymbolicName="MDIS_BinarySchema" DataType="ByteString">
    <DisplayName>Opc.MDIS</DisplayName>
    <References>
      <Reference ReferenceType="HasProperty">ns=1;i=376</Reference>
      <Reference ReferenceType="HasProperty">ns=1;i=15002</Reference>
      <Reference ReferenceType="HasComponent">ns=1;i=1485</Reference>
      <Reference ReferenceType="HasComponent" IsForward="false">i=93</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=72</Reference>
    </References>
    <Value>
      <ByteString xmlns="http://opcfoundation.org/UA/2008/02/Types.xsd">PG9wYzpUeXBlRGljdGlvbmFyeQ0KICB4bWxuczpvcGM9Imh0dHA6Ly9vcGNmb3VuZGF0aW9uLm9y
Zy9CaW5hcnlTY2hlbWEvIg0KICB4bWxuczp4c2k9Imh0dHA6Ly93d3cudzMub3JnLzIwMDEvWE1M
U2NoZW1hLWluc3RhbmNlIg0KICB4bWxuczp1YT0iaHR0cDovL29wY2ZvdW5kYXRpb24ub3JnL1VB
LyINCiAgeG1sbnM6dG5zPSJodHRwOi8vb3BjZm91bmRhdGlvbi5vcmcvVUEvTURJUyINCiAgRGVm
YXVsdEJ5dGVPcmRlcj0iTGl0dGxlRW5kaWFuIg0KICBUYXJnZXROYW1lc3BhY2U9Imh0dHA6Ly9v
cGNmb3VuZGF0aW9uLm9yZy9VQS9NRElTIg0KPg0KICA8b3BjOkltcG9ydCBOYW1lc3BhY2U9Imh0
dHA6Ly9vcGNmb3VuZGF0aW9uLm9yZy9VQS8iIExvY2F0aW9uPSJPcGMuVWEuQmluYXJ5U2NoZW1h
LmJzZCIvPg0KDQogIDxvcGM6RW51bWVyYXRlZFR5cGUgTmFtZT0iU2lnbmF0dXJlU3RhdHVzRW51
bSIgTGVuZ3RoSW5CaXRzPSIzMiI+DQogICAgPG9wYzpFbnVtZXJhdGVkVmFsdWUgTmFtZT0iTm90
QXZhaWxhYmxlIiBWYWx1ZT0iMSIgLz4NCiAgICA8b3BjOkVudW1lcmF0ZWRWYWx1ZSBOYW1lPSJD
b21wbGV0ZWQiIFZhbHVlPSIyIiAvPg0KICAgIDxvcGM6RW51bWVyYXRlZFZhbHVlIE5hbWU9IkZh
aWxlZCIgVmFsdWU9IjQiIC8+DQogIDwvb3BjOkVudW1lcmF0ZWRUeXBlPg0KDQogIDxvcGM6RW51
bWVyYXRlZFR5cGUgTmFtZT0iU2V0Q2FsY3VsYXRlZFBvc2l0aW9uRW51bSIgTGVuZ3RoSW5CaXRz
PSIzMiI+DQogICAgPG9wYzpFbnVtZXJhdGVkVmFsdWUgTmFtZT0iSW5pdGlhbCIgVmFsdWU9IjAi
```