OPCFoundation/UA-.NETStandard #699: Find the value MaxSubscriptionCount of using a client

bitbonk, 2019-04-11:
The application configuration of a OPC UA server allows me to [configure a MaxSubscriptionCount](https://github.com/OPCFoundation/UA-.NETStandard/blob/d007a41d150c59c65d5424552a6b1ad08b0e6541/SampleApplications/Workshop/DataAccess/Server/DataAccessServer.Config.xml#L132).

How can I get this server value from an OPC UA Client (e.g. throw some server diagnostics item in the address space, throw the client `Session` or `Endpoint` classes)?

AlinMoldovean, 2019-04-12:
@bitbonk ,

There is no standard defined for exposing this value in the server's address space.

bitbonk, 2019-04-12:
If not through the address space, maybe through the C# API?

AlinMoldovean, 2019-04-12:
@bitbonk ,

You can expose that parameter in a custom variable node of the server.
So the clients can read it.
But there is no standard way to expose it defined by UA Specs.

closed 2019-04-19 by AlinMoldovean