OPCFoundation/UA-.NETStandard #705: Node Id  status: Bad Out of service

asgaralim, 2019-04-16:
Repeat steps:
1) connect OPC UA COM Wrapper to OPC Classic server ( ClearSCADA using Thirdparty DA interface)
2) Connect OPC UA Client to COM Wrapper and browse the tags.

All Tag.Value gives Bad of of service status. Could you please let me know in which case COM wrapper return the "BadOutOfService". Please find below screenshot:

[image]

Thanks

AlinMoldovean, 2019-04-17:
@asgaralim ,

You need to make a debug inside the code against that DA server.
There are multiple places where that code can be generated.
It can be due to null return values in `Da30ReadItemValues `or `Da20ReadItemValues`.
Or the DA server returns quality code `OPC_QUALITY_OUT_OF_SERVICE `or `OPC_QUALITY_LAST_KNOWN`

asgaralim, 2019-04-23:
The clearscada support DA 2.0 interface.  I modified the code to use Da20ReadItemsValues.  
I created new user in clearscada and given all permission to access the tags/items.  using UA client (.Net Standard code) i am able to connect to the clearscada with user credentials and SignAndEncrypt:Basic256Sha256:Binary .  When i try to the view read the tag value it return  BadOutOfService. I debugged the code.  The method returns case ResultIds.E_ACCESSDENIED: { return StatusCodes.BadOutOfService; }. 

Note:- Interesting point here is that clearscada has build user "guest user" Assign this user to tag/items and provide all security permission (as defined for new user). Now try to read the tag value. It return the value as GOOD. 

Could you please let me what could be the issue. Please help me.

 private static ServiceResult MapReadErrorToServiceResult(
            int error,
            DiagnosticsMasks diagnosticsMasks)
        {
            if (error == 0)
            {
                return ServiceResult.Good;
            }

            switch (error)
            {
                case ResultIds.E_OUTOFMEMORY: { return StatusCodes.BadOutOfMemory; }
                case ResultIds.E_INVALIDHANDLE: { return StatusCodes.BadNodeIdUnknown; }
                case ResultIds.E_BADRIGHTS: { return StatusCodes.BadNotReadable; }
                case ResultIds.E_UNKNOWNITEMID: { return StatusCodes.BadNodeIdUnknown; }
                case ResultIds.E_INVALIDITEMID: { return StatusCodes.BadNodeIdInvalid; }
                case ResultIds.E_INVALID_PID: { return StatusCodes.BadNodeIdInvalid; }
                case ResultIds.E_ACCESSDENIED: { return StatusCodes.BadOutOfService; }
            }

            return StatusCodes.BadUnexpectedError;
        }

asgaralim, 2019-04-28:
I further debugged the code, While reading item value, it call the Da20ReadItemValues () At  ApplyChanges which finally calls the GetItemResults(). Here i get pError as negative value, which is mapped to E_ACCESSDENIED. I am unable to understand  what could be the reason.  Could you please help me.

asgaralim, 2019-05-03:
•	Connect to the server with PrivateSecurity (IOPCPrivate interface) and login with user credentials,with opc.tcp://vmwin2016dev:48400/UA/ComServerWrapper - [SignAndEncrypt:Basic256Sha256:Binary]
•	As the AddItems OPC method returns server handle =0 and errors as a negative number. This is finally mapped to the E_ACCESSDENIED which in-turn mapped to the BadOutOfservice.
•	See below logs:

5/3/2019 13:03:30.154 Da20ReadItemValues :Readrequest:ItemId=aws grop.test ,Value.value= ,ValueRequired=True  ,PropertyId=
5/3/2019 13:03:30.161 Group 4/6 Created(1) 0/250ms 0%
5/3/2019 13:03:30.166 AddItems - serverHandle 0: 
5/3/2019 13:03:30.170 AddItems - error -2147024891: 
5/3/2019 13:03:30.177 Group 4 AddItems(1/1) 0/250ms 0%
5/3/2019 13:03:30.181 Da20ReadItemValues :Readrequest:ItemId=aws grop.test ,Value.value= ,ValueRequired=True  ,PropertyId=
5/3/2019 13:03:30.185 Da20ReadItemValues-CheckforError :item.ItemId aws grop.test, item.ErrorId 0, item.ServerHandle 0, item.ClientHandle 4 
5/3/2019 13:03:30.190 Channel 5: SendResponse 314
5/3/2019 13:03:31.324 Channel 5: ProcessRequestMessage 315

[image]

asgaralim, 2019-05-05:
Analysis:
During Read and write operation, user identity credentials are not available to the UA COM Wrapper. This is due to the server context information is not correctly assigned to the UA COM Wrapper. The ComDAClient object was created from "SystemContext". The "SystemContext" does not contain identity information. Instead, the "context" server object contains the identity information.  To fix the issue the below code is modified to use "context" object. 

File Name : ..\ComIOP\Common\Client\Da\ComDaClientNodeManager.cs
Read Method  :  protected override void Write(ServerSystemContext context, ...........)
Write Method : protected override void Write(ServerSystemContext context,.............)            
Defect: ComDaClient client = system.SelectClient((ServerSystemContext)SystemContext, false);
Resolution: ComDaClient client = system.SelectClient((ServerSystemContext)context, false);

asgaralim, 2019-05-15:
> Hi @asgaralim <https://github.com/asgaralim> ,
>
> Could you provide a pull request containing the fix?
> We will test it and apply the changes.

Sure..., I will raise pull request.

closed 2019-09-26 by AlinMoldovean