OPCFoundation/UA-.NETStandard #708: Add support for https client TLS certificate authentication

ccasteyde, 2019-04-19:
The https binding doesn't check for TLS clients' certificate when the parameter AutoAcceptUntrustedCertificates is set to false.
This is a security issue, with configurations who do not use User authentication.
Implementing TLS clients' certificate validation requires modifying HttpsListener.cs to set Security.Transport.ClientCredentialType to HttpClientCredentialType.Certificate based on value of AutoAcceptUntrustedCertificates.
To be usefull and not trust all installed CAs in the Windows certificate store, use of custom validator should also be implemented via "m_host.Credentials.ClientCertificate.Authentication.CustomCertificateValidator = m_quotas.CertificateValidator;"
Please also note that the discovery interface of the server would require client certificate also, but that is not a problem for the client since it has to provide its TLS certificate anyway to connect.

ccasteyde, 2019-04-19:
See also #709

AlinMoldovean, 2019-04-24:
@ccasteyde ,

Similar with #692 
Specification says at Part6 7.4.1:

> HTTPS allows Clients to have certificates; however, they are not required by the HTTPS
transport. **A Server shall allow Clients to connect without providing a Certificate** during
negotiation of the HTTPS connection.

So if an application using HTTPS requires application certificate authentication, the validation should occur at _CreateSession_ service call instead of _CreateSecureChannel_.

closed 2019-05-22 by ccasteyde

ccasteyde, 2019-07-08:
We actually did some stack modifications in order to expose application validation of TLS client certificates. This requires adding options to require clients to provide a certificate first, then an option to check it with custom validator.
1. ApplicationConfiguration.cs, add two options:

`

        // BEGIN 20/05/2019 - github 708
        private bool m_RequireTLSCertificate;
        public bool RequireTLSCertificate
        {
            get { return m_RequireTLSCertificate; }
            set { m_RequireTLSCertificate = value; }
        }

        private bool m_UseCustomValidator;
        public bool UseCustomValidator
        {
            get { return m_UseCustomValidator; }
            set { m_UseCustomValidator = value; }
        }
        // END`

2. Use these in Core/Stack/Server/ServerBase.cs:

`

                    ITransportListener listener = new Opc.Ua.Bindings.UaHttpsChannelListener(
                        // BEGIN 18/04/2019 - Issue Github .NET Standard #708
                        configuration.SecurityConfiguration.AutoAcceptUntrustedCertificates || configuration.SecurityConfiguration.UseCustomValidator,
                        configuration.SecurityConfiguration.RequireTLSCertificate
                        // END
                        );

                    listener.Open(
                       uri.Uri,
                       settings,
                       GetEndpointInstance(this));`

3. Require client TLS certificate in channel (Core/Stack/Https/HttpsLiscener.cs):

`

    public partial class UaHttpsChannelListener : IDisposable, ITransportListener
    {
        // BEGIN 18/04/2019 - Issue Github .NET Standard #708
        bool m_bUseCustomValidator;
        bool m_bRequireClientCertificate;
        // END

        #region Constructors
        /// <summary>
        /// Initializes a new instance of the <see cref="UaTcpChannelListener"/> class.
        /// </summary>
        public UaHttpsChannelListener(
            // BEGIN 18/04/2019 - Issue Github .NET Standard #708
            bool bUseCustomValidator, bool bRequireClientCertificate
            // END
            )
        {
            // BEGIN 18/04/2019 - Issue Github .NET Standard #708
            m_bUseCustomValidator = bUseCustomValidator;
            m_bRequireClientCertificate = bRequireClientCertificate;
            // END
        }
...
        public void Start()
        {
            lock (m_lock)
            {
                UriBuilder root = new UriBuilder(m_uri);

[... 91 lines cut ...]
            // throw if rejected.
            if (!accept)
            {
                // write the invalid certificate to a directory if specified.
                lock (m_lock)
                {
                    Utils.Trace((int)Utils.TraceMasks.Error, "Certificate '{0}' rejected. Reason={1}", certificate.Subject, statusCode);

                    if (m_rejectedCertificateStore != null)
                    {
                        Utils.Trace((int)Utils.TraceMasks.Error, "Writing rejected certificate to directory: {0}", m_rejectedCertificateStore);
                        SaveCertificate(certificate);
                    }
                }

                throw new ServiceResultException(statusCode);
            }

            // add to list of peers.
            lock (m_lock)
            {
                m_validatedCertificates[certificate.Thumbprint] = certificate;
            }
            // END
        }`