OPCFoundation/UA-.NETStandard #709: .NET Standard Clients cannot connect to https servers that require TLS client authentication

ccasteyde, 2019-04-19:
The clients have to call GetEndPoints to get the available endpoints, but the DiscoveryClient class doesn't allow the client to provide its certificate (neither DiscoveryChannel). Therefore it is impossible to connect on https to a OPC UA server that requires the client to provide a TLS certificate.
Please also see report #708.

ccasteyde, 2019-04-19:
HttpsTransportChannel does not set up the client's certificate either, nor the Create method of Session.cs in the sample.

AlinMoldovean, 2019-04-24:
@ccasteyde ,

According to the latest specification UA servers shall allow clients to connect without providing a certificate during the negotiation of the HTTPS connection, as described in #708
So the server should allow clients that do not provide a client certificate at GetEndpoints.

For RegisterServer and RegisterServer2 services it is stated in Part4 5.4 that they cannot be used  with HTTPS:

> This Service can only be invoked via SecureChannels that support Client authentication (i.e.
HTTPS cannot be used to call this Service).

ccasteyde, 2019-05-22:
Hello
Right, the spec says https servers should always accept client that do not provide TLS certificate, therefore this can be considered as a "non issue".
However servers can be configured to require a client certificate, as an extension, and for interoperability it should be better if the client stack could send its certificate.
I've made no assumption on RegisterServer (called on LDS by the server), but on  "GetEndPoints" (called by client to get the available endpoints). If a server use https, it's difficult to support it on a per method basis, and GetEndPoints will use the same configuration than CreateSession on most cases.
I have some code where Application Certificate is used to configure the channel, which works with servers that requires client certificates and servers that do not require them.
I'll post it later.

ccasteyde, 2019-07-08:
Here is the code we finally used.

1. Need to give the DiscoveryClient class the application configuration (use decided to use application certificate as TLS client certificate for sake of configuration simplicity). Also give the certificate to use to DiscoveryChannel class:
  Core/Stack/Client/DiscoveryClient.cs:
` // New class factory in DiscoveryClient:
        public static DiscoveryClient Create(Uri discoveryUrl, EndpointConfiguration configuration, ApplicationConfiguration applicationConfiguration)
        {
            System.Security.Cryptography.X509Certificates.X509Certificate2 certificate = applicationConfiguration.SecurityConfiguration.ApplicationCertificate.Certificate;
            ITransportChannel channel = DiscoveryChannel.Create(discoveryUrl, configuration, new ServiceMessageContext(), certificate);
            return new DiscoveryClient(channel);
        }

...

// Extra parameters in DiscoveryChannel constructors:
public static ITransportChannel Create(
            Uri discoveryUrl,
            EndpointConfiguration endpointConfiguration,
            ServiceMessageContext messageContext
            // BEGIN 19/04/2019- Issue Github .Net Standard #709
            , System.Security.Cryptography.X509Certificates.X509Certificate2 certificate = null
            // END
            )
        {
...
ITransportChannel channel = CreateUaBinaryChannel(
                null,
                endpoint,
                endpointConfiguration,
                // BEGIN 19/04/2019- Issue Github .Net Standard #709
                // (System.Security.Cryptography.X509Certificates.X509Certificate2)null,
                certificate,
                // END
                messageContext);
`

2. Use the certificate in BeginSendRequest method of HttpsTransportChannel class (HttpsTransportChannel.cs):

`
           HttpWebRequest webRequest = (HttpWebRequest)HttpWebRequest.Create(m_url.ToString());
            // BEGIN 19/04/2019- Issue Github .Net Standard #709
            if (m_settings.ClientCertificate != null)
                webRequest.ClientCertificates.Add(m_settings.ClientCertificate);
            // END
            webRequest.Method = "POST";
`

3. Allow use of https with security None and/or no TLS client certificate in SampleLibraries (Client/Session.cs), method Create:
` 
           // checks the domains in the certificate.
            if (checkDomain && endpoint.Description.ServerCertificate != null && endpoint.Description.ServerCertificate.Length > 0)
            {
                CheckCertificateDomain(endpoint);
            }

            X509Certificate2 clientCertificate = null;
            X509Certificate2Collection clientCertificateChain = null;

            // BEGIN 19/04/2019- Issue Github .Net Standard #709
            bool useHttps = false;
            switch (endpointDescription.TransportProfileUri)
            {
                case Profiles.HttpsXmlTransport:
                case Profiles.HttpsBinaryTransport:
                case Profiles.HttpsXmlOrBinaryTransport:
                    {
                        useHttps = true;
                        break;
                    }
            }
            if (endpointDescription.SecurityPolicyUri != SecurityPolicies.None || useHttps)
            //if (endpointDescription.SecurityPolicyUri != SecurityPolicies.None)
            // END
            {
                if (configuration.SecurityConfiguration.ApplicationCertificate == null)
                {
                    // BEGIN 19/04/2019- Issue Github .Net Standard #709
                    if (!useHttps)
                    // END
                        throw ServiceResultException.Create(StatusCodes.BadConfigurationError, "ApplicationCertificate must be specified.");
                }

                clientCertificate = configuration.SecurityConfiguration.ApplicationCertificate.Find(true);

                if (clientCertificate == null)
                {
                    // BEGIN 19/04/2019- Issue Github .Net Standard #709
                    if (!useHttps)
                    // END
                        throw ServiceResultException.Create(StatusCodes.BadConfigurationError, "ApplicationCertificate cannot be found.");
                }
                else
                {
...
`

commit a1e26bf8 (2019-10-02): Interoperability fixes for HTTPS (#814)

* Fixes #709 - Allow HTTPS clients to send the certificate for servers that require TLS client authentication

* Fixes #692 - HTTPS Server listener - Use the first configured policy for HTTPS
closed 2019-10-02 by AlinMoldovean