OPCFoundation/UA-.NETStandard #710: Question about renewing of server certificates

Ckath, 2019-04-19:
I'm currently looking into how the deal with certs that would work in production, and theres one thing I cant really see the workflow of. What is supposed to happen when the server certificate expires?

From what I can see the server generates its own certificates when they don't exist, which is great. I'd then trust this with my clients and add the client's certificates to the server's trusted store. This is all still going great, but then I imagine what happens when the certificate of the server expires, I'd hoped it would renew it, or generate a new one, but this is not the case. Testing this by changing around dates and the quickstart reference server **it errors out when starting the server with an expired certificate**.

My question here would be how I should set this up for long term usage, restarting the server for it to work wouldn't be a huge dealbreaker. Should I be deleting the self signed certs and letting the server recreate them? Should I handle it similar to a websites certs and use a third party solution to renew certificates? (this option would be bit complex for what the opc ua server will run on)  Are there ways to more gracefully set this up from within the server implementation?

fchaxel, 2019-04-20:
Hi,

Looks at #678

Bye.

Ckath, 2019-04-20:
ah i see that goes over basically the same point, thanks. if I run into anything different I'll see then.

closed 2019-04-20 by Ckath