OPCFoundation/UA-.NETStandard #711: Windows Authentication (Active Directory)

vabrutski, 2019-04-23:
Hello, 

is there a built in way to authenticate client on server using windows authentication? If client already authenticated then I expect that it will be automatically authenticated on server using it's credentials(without need to enter login and password again)

teozfrank, 2019-04-24:
No this is not part of the standard authenticating methods. Maybe in the future LDAP may be supported possibly?

vabrutski, 2019-04-24:
> No this is not part of the standard authenticating methods. Maybe in the future LDAP may be supported possibly?

That's very strange. As long as OPC UA uses WCF behind. And WCF has native support of windows authentication(wsBinnding and httpBinding).

teozfrank, 2019-04-25:
As part of the OPC UA standard, there are 3 types of client autentication. Anonymous, username & password or certificate. Any other types you would have to implement yourself both on the client and server side. The downside of this is that this type of custom authentication would be restricted by the use of your own custom client and server which really wouldn't make it a "standard".

PaulHunkar, 2019-04-25:
In version 1.04 OAuth2  and JWT have been added for User authentication.  This may cover what you are looking for.

AlinMoldovean, 2019-10-16:
Sample code for OAuth2 authentication with JWT added on "prototyping-OAuth2" branch

closed 2019-10-16 by AlinMoldovean