OPCFoundation/UA-.NETStandard #712: securityPolicyUri is not null checked when seting requireEncryption value

claudiocampeggi, 2019-04-24:
Hi,
I'm in the process of switching from the UA legacy libraries to the Standard ones.
I connect to a customer's Siemens S7-1500 plc already in the production environment which has no encryption, no authentication and an expired certificate (I know it's unsafe, but I can't do anything about it).

The first attempt to use the Standard libraries failed because the certificate validation throws an `BadCertificateTimeInvalid` error; I believe it should not try to validate the certificates so I investigated the connection process.

In the latest version (as of today) of the Standard libraries I noticed that in file `Session.cs` [lines 2101-2105](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/SampleApplications/SDK/Opc.Ua.Client/Session.cs#L2101)

the variable `bool requireEncryption` is true for my configuration; it is true because `identityPolicy.SecurityPolicyUri` is null and so line 2104 makes it true.

In turn, `identityPolicy.SecurityPolicyUri` is null because the call at [line 2084](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/SampleApplications/SDK/Opc.Ua.Client/Session.cs#L2084) is 
`m_endpoint.Description.FindUserTokenPolicy(UserTokenType.Anonymous, null);`

I believe that a null value should equals to `SecurityPolicy.None` so both statements at [line 2101](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/SampleApplications/SDK/Opc.Ua.Client/Session.cs#L2101) and [line 2104](https://github.com/OPCFoundation/UA-.NETStandard/blob/master/SampleApplications/SDK/Opc.Ua.Client/Session.cs#L2104) should check for null or empty values.

Below my suggested change:
`bool requireEncryption = !string.IsNullOrEmpty(securityPolicyUri) && securityPolicyUri != SecurityPolicies.None;
            if (!requireEncryption)
            {
                requireEncryption = !string.IsNullOrEmpty(identityPolicy.SecurityPolicyUri) && identityPolicy.SecurityPolicyUri != SecurityPolicies.None;
            }`

thank you
Best regards

DummyRick, 2019-07-05:
Hi,

I'm trying to comunicate with the same PLC and the same issue appear also to me.
Fixed using @claudiocampeggi code.

Please, let me know if the code will be accepted or not. I don't want to have custom modifications to these libraries.

AlinMoldovean, 2019-07-22:
@claudiocampeggi , @DummyRick ,

We are waiting from you a valid pull request with the proposed code change.
We will review and test it.

closed 2019-11-21 by AlinMoldovean