OPCFoundation/UA-.NETStandard #714: Decrypt the message encodeable

Sun-Dordje, 2019-04-25:
Hello the community,

I work on a NIDS for OPC UA. I use the tool Zeek to catch and parse opc ua's packets.
I have install the framework open62541 to have some client/server and dump the trafic to make some test.
When the security mode is None, there is no problem to have each fields.

When the security mode is basic256sha256 i have a problem, i want to decrypt the "Message : Encodeable Object"
but i dont know how it is really encrypted, there is my opc ua trafic on a pcap file:

[image]

I see the security policy define in the part 7: So the OpenSecureChannel use the AsymmetricEncryptionAlgorithm_RSA-OAEP-SHA1
when the security mode is basic256sha256 (http://opcfoundation.org/UA/SecurityPolicy#Basic256Sha256).
I have implement it with python.

There is my outputs from my NIDS when the OpenSecureChannel "basic256sha256" is catch.

[image]

When i catch the sendercertif, i build it like a hexa-file and i have the right certificat.

[image]

When i catch the encrypted message (wich is exactly the same as the pcap file), i want to decrypt it with RSAES-OAEP-DECRYPT and with the good private key.
But I dont know how can i do because to decrypt a message, the message must have 256 bytes. But here i have 504 bytes.
Maybe there is a padding or anything else but i tried to decrypt each 256 bytes block from the message i had catch, but no success.

If someone can help me it is very cool, sorry for my english.

thank you
Seunam

AlinMoldovean, 2019-04-25:
Hi @Sun-Dordje,

When using the security mode SignAndEncrypt, the main intent is that the communication cannot be decoded if somebody is able to intercept it. e.g. using a sniffer tool like WireShark.
The public certificates exchanged during connection establishment cannot be used to decrypt the communication.

In order to be able to decrypt the data you need the private key of the certificates. And that is not exchanged by the applications.

You can take a look at the implementation of `RsaUtils.Encrypt()/Decrypt()` methods of this stack.

Sun-Dordje, 2019-04-26:
Hi @AlinMoldovean, Thanks to answer.

>The public certificates exchanged during connection establishment cannot be used to decrypt the communication.

I know this, but i intercept the certificate only for looking if my NIDS capture is good.

>In order to be able to decrypt the data you need the private key of the certificates. And that is not exchanged by the applications.

I have the certificate's private key, and i have implement the AsymmetricEncryptionAlgorithm_RSA-OAEP-SHA1 from the rfc3447 in python to try to decrypt what i catch but for each crypted message it is not a multiple of 256 bytes, so i dont know how to decrypt.

>You can take a look at the implementation of RsaUtils.Encrypt()/Decrypt() methods of this stack.

Ok i will see it.