OPCFoundation/UA-.NETStandard #717: BadInvalidTimestamp while server time is ahead of client

Ckath, 2019-05-02:
I've created an opc ua server for our company based on the reference server and its been working out for the most, but today with some integration testing we ran into a weird issue: if the server time is **ahead** of the client machines they return 'BadInvalidTimestamp', however if the server time is **behind** of client time it doesn't matter and all you get is a warning on connect.

This happens on both the softing's DataFeed client and unified automation's UaExpert, the errors also don't look like the server is choking on something, but rather the clients aren't able to handle this situation:
[image]
[image]

We have no way to guarantee the remote machines that would be running the OPC UA Server are set to a sensible time, is there anything I can change in the server that would keep this from happening to the clients?  And if anyone can explain to me why this happens only when the server time is ahead of client time but its fine otherwise, that would be great as well. 

Workarounds on client side aren't ideal as the client is left up to our clients, we only provide the server side.

Cheers, Ck

Ckath, 2019-05-09:
This certainly looks promising, are there any issues that could come up by using the server with an extremely big *MaxRequestAge*, or even bypassing the check in completely?

AlinMoldovean, 2019-05-09:
@Ckath ,

According to the specification, the client and server clocks should be in sync.
That is the reason of that check.
Increasing that time tolerance should not create issues.

closed 2019-05-23 by AlinMoldovean