OPCFoundation/UA-.NETStandard #718: BadSecurityChecksFailed 'Error received from remote host

asgaralim, 2019-05-03:


asgaralim, 2019-05-03:
Good Morning...!

I see below log messages which logged repeatly. Could you please let me know what these error message for. Does these error messages has any impact on item quality status and client connection which uses IOPC private security.

5/3/2019 10:12:13.443 Channel 4: ProcessRequestMessage 105
5/3/2019 10:12:13.452 Channel 4: SendResponse 105
5/3/2019 10:12:18.443 Channel 4: ProcessRequestMessage 106
5/3/2019 10:12:18.450 Channel 4: SendResponse 106
5/3/2019 10:12:23.443 Channel 4: ProcessRequestMessage 107
5/3/2019 10:12:23.452 Channel 4: SendResponse 107
5/3/2019 10:12:28.444 Channel 4: ProcessRequestMessage 108
5/3/2019 10:12:28.451 Channel 4: SendResponse 108
5/3/2019 10:12:33.447 Channel 4: ProcessRequestMessage 109
5/3/2019 10:12:33.457 Channel 4: SendResponse 109
5/3/2019 10:12:38.447 Channel 4: ProcessRequestMessage 110
5/3/2019 10:12:38.454 Channel 4: SendResponse 110
5/3/2019 10:12:39.369 Channel 0 in Connecting state.
5/3/2019 10:12:39.382 Channel 0: SendHelloMessage()
5/3/2019 10:12:39.388 Channel 0: ProcessAcknowledgeMessage()
5/3/2019 10:12:39.395 Channel 0 in Opening state.
5/3/2019 10:12:39.402 Token #0 created. CreatedAt = 09:12:39.402 . Lifetime = 3600000
5/3/2019 10:12:39.422 Channel 0: ProcessErrorMessage(BadSecurityChecksFailed 'Error received from remote host: ')
5/3/2019 10:12:39.430 ***EXCEPTION*** BadSecurityChecksFailed 'Error received from remote host: '
5/3/2019 10:12:39.463 ***EXCEPTION*** BadSecurityChecksFailed 'Error establishing a connection: Error received from remote host: '
5/3/2019 10:12:39.477 Channel 0 in Closed state.
5/3/2019 10:12:39.487 CLIENTCHANNEL SOCKET CLOSED: 02425494, ChannelId=0
5/3/2019 10:12:39.494 RegisterServer2 failed for at: opc.tcp://localhost:4840/. Exception=Error establishing a connection: Error received from remote host: 
5/3/2019 10:12:39.500 Channel 0 in Connecting state.
5/3/2019 10:12:40.056 Channel 0: SendHelloMessage()
5/3/2019 10:12:40.063 Channel 0: ProcessAcknowledgeMessage()
5/3/2019 10:12:40.069 Channel 0 in Opening state.
5/3/2019 10:12:40.074 Token #0 created. CreatedAt = 09:12:40.074 . Lifetime = 3600000
5/3/2019 10:12:40.093 Channel 0: ProcessErrorMessage(BadSecurityChecksFailed 'Error received from remote host: ')
5/3/2019 10:12:40.099 ***EXCEPTION*** BadSecurityChecksFailed 'Error received from remote host: '
5/3/2019 10:12:40.133 ***EXCEPTION*** BadSecurityChecksFailed 'Error establishing a connection: Error received from remote host: '
5/3/2019 10:12:40.169 Channel 0 in Closed state.
5/3/2019 10:12:40.197 CLIENTCHANNEL SOCKET CLOSED: 01F10179, ChannelId=0
5/3/2019 10:12:40.208 RegisterServer failed for at: opc.tcp://localhost:4840/. Exception=Error establishing a connection: Error received from remote host: 
5/3/2019 10:12:40.218 Register server failed. Trying again in 30000 ms
5/3/2019 10:12:43.447 Channel 4: ProcessRequestMessage 111
5/3/2019 10:12:43.454 Channel 4: SendResponse 111
5/3/2019 10:12:48.449 Channel 4: ProcessRequestMessage 112
5/3/2019 10:12:48.459 Channel 4: SendResponse 112
5/3/2019 10:12:53.449 Channel 4: ProcessRequestMessage 113
5/3/2019 10:12:53.457 Channel 4: SendResponse 113
5/3/2019 10:12:58.450 Channel 4: ProcessRequestMessage 114
5/3/2019 10:12:58.460 Channel 4: SendResponse 114
5/3/2019 10:13:03.451 Channel 4: ProcessRequestMessage 115
5/3/2019 10:13:03.459 Channel 4: SendResponse 115
5/3/2019 10:13:08.452 Channel 4: ProcessRequestMessage 116
5/3/2019 10:13:08.461 Channel 4: SendResponse 116
5/3/2019 10:13:10.283 Channel 0 in Connecting state.
5/3/2019 10:13:10.304 Channel 0: SendHelloMessage()
5/3/2019 10:13:10.314 Channel 0: ProcessAcknowledgeMessage()
5/3/2019 10:13:10.325 Channel 0 in Opening state.
5/3/2019 10:13:10.333 Token #0 created. CreatedAt = 09:13:10.333 . Lifetime = 3600000
5/3/2019 10:13:10.351 Channel 0: ProcessErrorMessage(BadSecurityChecksFailed 'Error received from remote host: ')
5/3/2019 10:13:10.359 ***EXCEPTION*** BadSecurityChecksFailed 'Error received from remote host: '
5/3/2019 10:13:10.375 Channel 0 in Closed state.
5/3/2019 10:13:10.376 ***EXCEPTION*** BadSecurityChecksFailed 'Error establishing a connection: Error received from remote host: '
5/3/2019 10:13:10.382 CLIENTCHANNEL SOCKET CLOSED: 0370F030, ChannelId=0
5/3/2019 10:13:10.397 RegisterServer2 failed for at: opc.tcp://localhost:4840/. Exception=Error establishing a connection: Error received from remote host: 
5/3/2019 10:13:10.403 Channel 0 in Connecting state.
5/3/2019 10:13:10.409 Channel 0: SendHelloMessage()
5/3/2019 10:13:10.415 Channel 0: ProcessAcknowledgeMessage()
5/3/2019 10:13:10.420 Channel 0 in Opening state.
5/3/2019 10:13:10.426 Token #0 created. CreatedAt = 09:13:10.426 . Lifetime = 3600000
5/3/2019 10:13:10.444 Channel 0: ProcessErrorMessage(BadSecurityChecksFailed 'Error received from remote host: ')
5/3/2019 10:13:10.450 ***EXCEPTION*** BadSecurityChecksFailed 'Error received from remote host: '
5/3/2019 10:13:10.481 ***EXCEPTION*** BadSecurityChecksFailed 'Error establishing a connection: Error received from remote host: '
5/3/2019 10:13:10.526 Channel 0 in Closed state.
5/3/2019 10:13:10.532 CLIENTCHANNEL SOCKET CLOSED: 01589D1A, ChannelId=0
5/3/2019 10:13:10.538 RegisterServer failed for at: opc.tcp://localhost:4840/. Exception=Error establishing a connection: Error received from remote host: 
5/3/2019 10:13:10.544 Register server failed. Trying again in 30000 ms
5/3/2019 10:13:13.453 Channel 4: ProcessRequestMessage 117
5/3/2019 10:13:13.463 Channel 4: SendResponse 117
[... cut ...]
5/3/2019 10:13:18.461 Channel 4: SendResponse 118
5/3/2019 10:13:23.455 Channel 4: ProcessRequestMessage 119
5/3/2019 10:13:23.462 Channel 4: SendResponse 119
5/3/2019 10:13:28.455 Channel 4: ProcessRequestMessage 120
5/3/2019 10:13:28.463 Channel 4: SendResponse 120
5/3/2019 10:13:33.456 Channel 4: ProcessRequestMessage 121
5/3/2019 10:13:33.465 Channel 4: SendResponse 121
5/3/2019 10:13:38.457 Channel 4: ProcessRequestMessage 122
5/3/2019 10:13:38.470 Channel 4: SendResponse 122
5/3/2019 10:13:40.600 Channel 0 in Connecting state.

Zigby, 2019-05-06:
It looks like you have a server that tries to register with a LDS (local discovery server) which is by default using port 4840. Check the Config.xml file for a <RegistrationEndpoint> entry. You can remove the <RegistrationEndpoint> in case you do not want to register your server with the LDS, which you may not have installed, or certificates are not mutually trusted.

closed 2019-05-23 by AlinMoldovean