OPCFoundation/UA-.NETStandard #719: CNC System

jixiedaima, 2019-05-06:
I'm doing data acquisition here. Can you tell me which CNC systems are supported now?

MD-V, 2019-05-06:
Hello @jixiedaima ,

can you elaborate your question a little more?
You have posted in the issue section for the OPC UA .NET developer SDK and not in a general OPC UA question forum.
If you want to know if the manufacturer of your CNC has an OPC UA server please contact the manufacturer directly.

jixiedaima, 2019-05-07:
Okay， thank you！

cpipero, 2019-05-07:
Also you might want to look at two related intiative where you can find a number of CNC equipment vendors supporting OPC UA:
- MTConnect, info at [https://www.mtconnect.org/opc-ua-companion-specification](https://www.mtconnect.org/opc-ua-companion-specification)
- Umati, info at [https://en.industryarena.com/umati/documents/umati-flyer--4759.html](https://en.industryarena.com/umati/documents/umati-flyer--4759.html)

jixiedaima, 2019-05-08:
Thanks a lot.

closed 2019-05-23 by AlinMoldovean