OPCFoundation/UA-.NETStandard #723: Could not load nodes from resource (Opc.Ua.Sdk.ServiceResultException)

NakWarsi, 2019-05-22:
I am trying to load NodeSets from a binary file, I tried a few formats of binary files (.uanodes, .ua, .tt2pro etc) to make sure that if there is not any issue with file format but every time I am getting the same exception at the time of uploading resource.
Code:
```
          NodeStateCollection predefinedNodes = new NodeStateCollection();
                predefinedNodes.LoadFromBinaryResource(context,
                    "umati.tt2pro",
                    typeof(ServerNodeManager_Binary).GetTypeInfo().Assembly,
                    true);
                return predefinedNodes;
```
Exception:
```
Opc.Ua.Sdk.ServiceResultException
  HResult=0x80131500
  Message=Could not load nodes from resource: umati.tt2pro
  Source=Opc.Ua.Sdk.Core
  StackTrace:
   at Opc.Ua.Sdk.NodeStateCollection.LoadFromBinaryResource(ISystemContext context, String resourcePath, Assembly assembly, Boolean updateTables)
   at abc.OPCUA.Server.ServerNodeManager_Binary.LoadPredefinedNodes(ISystemContext context) in C:\Users\Wars_Nau\source\repos\CycloidGearClosedLoop\src\UAServer\NodeManagers\ServerNodeManager_Binary.cs:line 66
   at abc.OPCUA.Server.ServerNodeManager_Binary.CreateAddressSpace(IDictionary`2 externalReferences) in C:\Users\Wars_Nau\source\repos\CycloidGearClosedLoop\src\UAServer\NodeManagers\ServerNodeManager_Binary.cs:line 74
   at Opc.Ua.Sdk.Server.MasterNodeManager.Startup()

```

MD-V, 2019-05-22:
Hello @NakWarsi,

.tt2pro is a UnifiedAutomation UAModeler project file. To load nodes into the stack you'll need a NodeSet2 XML file (you can generate it via the UAModeler) or a uanodes file. 

Are you sure that you have tried it with a valid uanodes file? You could try to load a example file contained in this repo (e.g. [Quickstarts.Boiler.PredefinedNodes.uanodes](https://github.com/OPCFoundation/UA-.NETStandard/blob/1121363a4cea49c461516353ef412ab56c3fe5c7/SampleApplications/Workshop/Boiler/Server/Quickstarts.Boiler.PredefinedNodes.uanodes)) to verify that your code is working and afterwards run the same code with your uanodes file.

NakWarsi, 2019-05-22:
Yes i tried .uanodes as well  with the same project reference you have provided above but still i am getting the same issue.

MD-V, 2019-05-22:
According to the source code the file you want to load is not found in the embedded resources. ([see here](https://github.com/OPCFoundation/UA-.NETStandard/blob/f1cdabc034ce795e930658d5a021385a7815b826/Stack/Opc.Ua.Core/Stack/State/NodeStateCollection.cs#L446))

1. Please check if you have marked the uanodes file as embedded resource 
2. Please check if the name of the embedded resource is correct. (Note that the format is [assembly name].[resource name])

NakWarsi, 2019-05-22:
Thank You @MD-V

closed 2019-05-22 by NakWarsi