OPCFoundation/UA-.NETStandard #727: Exception when UA COM Server Wrapper is under much load

bjaben, 2019-05-23:
Sorry if I'm asking at the wrong place but I'm seeing the following error when running the UA COM Server Wrapper on a Windows 7 machine. The error starts showing up when I sample a few hundred nodes every 1 second using this [OPCPublisher](https://github.com/Azure/iot-edge-opc-publisher):

```
IOPCItemProperties.GetItemProperties ObjectDisposedException 'Cannot access a disposed object.
Object name: 'The COM server has been disposed.'
```
As this error does not happen on all Windows7 machines that I have tested I'm sure it's not related to the UA COM Server Wrapper itself but I thought this would be the best place to ask if someone has seen this error before or can provide some guidance on why the COM Server is being disposed under this load.

petrovickn, 2019-11-28:
Is there any news on this issue? We are facing the same problem in our project.

ezequielfalcon, 2020-03-05:
This is old, but something similar is happening to me.
We're using the Wrapper with a GE iFIX OPC DA server. If we use a standard OCP UA Client (like Integration Objects one) it works flawlessly, but I soon as I try to get a NodeID variable with `node-opcua` library it throws this same error.