OPCFoundation/UA-.NETStandard #729: Connection to server having self-signed certificate fails with chain incomplete

andyometa, 2019-05-29:
On a project (OPC UA client application) I'm working on we use a Beckhoff PLC device for testing purposes.  The OPC UA server on the PLC contains a self-signed certificate, issuer and subject fields are exactly the same.

With the SDK version 1.04.353.15 we still could connect properly without any issues.  However, after an upgrade to SDK version 1.4.355.26, suddenly an "Certificate chain validation incomplete" error message starts to pop up.

After some searching and debugging, I noticed that one particular part in the class ChainValidator, method InternalValidate seems to skip a check (lines 730 - 746):

```csharp
// check for chain status errors.
foreach (X509ChainStatus status in element.ChainElementStatus)
{
    ServiceResult result = CheckChainStatus(status, target, issuer, (ii != 0));

    if (ServiceResult.IsBad(result))
    {
        // check untrusted certificates.
        if (trustedCertificate == null)
        {
            throw new ServiceResultException(StatusCodes.BadSecurityChecksFailed);
        }

        throw new ServiceResultException(result);
    }
    chainStatusChecked = true;
}
```

The self-signed certificate doesn't have a X509ChainStatus struct which leads to the fact that the chainStatusChecked will never be set to true.
And as such, the following exception is always thrown (lines 774 - 781):

```csharp
if (!chainStatusChecked || chainIncomplete)
{
    throw ServiceResultException.Create(
        StatusCodes.BadCertificateChainIncomplete,
        "Certificate chain validation incomplete.\r\nSubjectName: {0}\r\nIssuerName: {1}",
        certificate.SubjectName.Name,
        certificate.IssuerName.Name);
}
```

I'd like to know whether this behaviour (in)correct, or a faulty self-signed certificate. Is there a kind of workaround, e.g. by setting the application configuration property SendCertificateChain to either true or false?

AlinMoldovean, 2019-05-31:
Hello @andyometa ,

This is something that should be fixed with commit 24e44dd3f3e55d53875ef2b3f5b0b2759bbb481c
Could you check if the issue replicates with the latest code from the master branch?

chandru9050, 2019-05-31:
Hello @AlinMoldovean 

I am facing this issue too, even I tried your fix suggested, I am getting the same problem"Certificate chain validation incomplete" for a self signed cert

andyometa, 2019-06-03:
@AlinMoldovean 

Thank you for the fix.
I'll check the new code and let you know.

andyometa, 2019-06-04:
@AlinMoldovean 

I'm afraid I've bad news.  Somehow I still keep getting the "Certificate chain validation incomplete" error message.
Hopefully I'm able to pinpoint the location through debugging one of the coming days if I can make the time.

chandru9050, 2019-06-04:
Hello @AlinMoldovean 

Yes I tried the same, same error,

AlinMoldovean, 2019-06-04:
This can happen if the certificate is issued by a CA i.e. the following check return _false_:
```
Utils.CompareDistinguishedName(certificate.Subject, certificate.Issuer);
```

@chandru9050 , @andyometa 
Could you check if the certificate has the same `Subject `and `Issuer `in order to be considered self signed by the validator.

lest123, 2019-06-05:
I'm facing the same issue with a self-signed cert from Kepserver.
I _think_ the problem is in Opc.Ua.Core.CertificateValidator.InternalValidate. If element.ChainElementStatus is empty chainStatusChecked is not set and an error is raised.
According to https://docs.microsoft.com/en-us/dotnet/api/system.security.cryptography.x509certificates.x509chain?view=netframework-4.8 chain.ChainStatus should be used for validation.

andyometa, 2019-06-05:
I checked the certificate to be sure, and yes the subject and issuer are exactly the same. So that is correct.

After debugging the code again, I notice that the exception is triggered at the same location like @lest123 says, class `CertificateValidator`, function `InternalValidate`.

The loop (`foreach (X509ChainStatus status in element.ChainElementStatus)`) will never execute because it seems that the ChainElementStatus is always null or empty for self-signed certificates and that results in the fact that the variable `chainStatusChecked` will never be set to true.

As such the following check will always be true, triggering the exception:

```csharp
if (!chainStatusChecked || chainIncomplete)
{
    throw ServiceResultException.Create(
        StatusCodes.BadCertificateChainIncomplete,
        "Certificate chain validation incomplete.\r\nSubjectName: {0}\r\nIssuerName: {1}",
        certificate.SubjectName.Name,
        certificate.IssuerName.Name);
}
```

chandru9050, 2019-06-06:
@AlinMoldovean  , yes I too checked, The subject and issuer are same, and I am to connect to the server with other UA Clients

commit 842c7e28 (2019-06-07): Merge pull request #738 from AlinMoldovean/master

Fixes #729. Connection with self-signed certificate fails.
commit 9b026522 (2019-06-07): Fixes #729. Connection with self-signed certificate fails with BadCertificateChainIncomplete.
closed 2019-06-07 by AlinMoldovean