OPCFoundation/UA-.NETStandard #730: BadTcpInternalError ConnectionReset

PhantoManresa, 2019-05-29:
Hi, I made an OPC UA client. When I connect my client to an OPC UA server everything seems to work fine at first, but after an aparently random amount of time I get the following `ServiceResultException`:

Result: `BadSecureChannelClosed`
InnerResult: `BadTcpInternalError`
Description: ConnectionReset

After that, the client reconnects to the server and the same keep happening, until for some reason it can no longer reconnect.
Can anyone tell me why I am having this error?
Thanks in advance

AlinMoldovean, 2019-05-31:
@PhantoManresa ,

The described error is normally generated in case the connection with the server is lost.
That can happen from multiple reasons, e.g. the server closes the session or becomes unresponsive, the network connection is broken.
You need to check also on the server side for this.

PhantoManresa, 2019-05-31:
I am connecting to the server provided by Siemen's Tia Portal. I don't know how to check the server side in this case. In this case we have the server and the client in the same machine, so network connection is not the issue. It could be that the server closes the session. We recently noted that we get first a `BadSessionIdInvalid` error. We have connected to Tia Portal's OPC UA server previously, but this never happened to us before.
Any idea why the server could be closing the session?
Thanks again.

AlinMoldovean, 2019-05-31:
_BadSessionIdInvalid_  response means that the server has disposed the session for some reason.
That can be due to client incatiinactivity, but the keep-alive mechanism should prevent that.
You need to check on the server side for more information.

PhantoManresa, 2019-05-31:
We even exchange information with the server every second, apart from the keep-alive mechanism. I guess we will try with the server side.
Thank you for the all help and attention.

closed 2019-05-31 by AlinMoldovean