OPCFoundation/UA-.NETStandard #732: [Question] How can I save trusted ICertificateStore as .pem file?

Never7Dice, 2019-05-29:
In application I have access to ICertificateStore object of trusted certificates, but the logic of another library requires me to have .pem truststore file.

What would be the best way to export trusted store to file in format like this?

```
Bag Attributes
    friendlyName: landoopfastdatadevca
    2.16.840.1.113894.746875.1.1: <Unsupported tag 6>
subject=C = Ankh-Morpork, O = Unseen University, OU = Library, CN = Landoop's Fast Data Dev Self Signed Certificate Authority

issuer=C = Ankh-Morpork, O = Unseen University, OU = Library, CN = Landoop's Fast Data Dev Self Signed Certificate Authority

-----BEGIN CERTIFICATE-----
MIID7jCCAtagAwIBAgIRAM3wpmyZO+uS5t9vutGEXgQwDQYJKoZIhvcNAQELBQAw
gYkxFTATBgNVBAYTDEFua2gtTW9ycG9yazEaMBgGA1UEChMRVW5zZWVuIFVuaXZl
cnNpdHkxEDAOBgNVBAsTB0xpYnJhcnkxQjBABgNVBAMTOUxhbmRvb3AncyBGYXN0
IERhdGEgRGV2IFNlbGYgU2lnbmVkIENlcnRpZmljYXRlIEF1dGhvcml0eTAeFw0x
OTA1MTgxNDAxNDZaFw0yMDA1MTcyMDAxNDZaMIGJMRUwEwYDVQQGEwxBbmtoLU1v
cnBvcmsxGjAYBgNVBAoTEVVuc2VlbiBVbml2ZXJzaXR5MRAwDgYDVQQLEwdMaWJy
YXJ5MUIwQAYDVQQDEzlMYW5kb29wJ3MgRmFzdCBEYXRhIERldiBTZWxmIFNpZ25l
ZCBDZXJ0aWZpY2F0ZSBBdXRob3JpdHkwggEiMA0GCSqGSIb3DQEBAQUAA4IBDwAw
ggEKAoIBAQC33aoq+F1EYC1bpFijL4INRACHJieqR9P2vJctIXpFzmImSD+KNMkt
JXGOmqc/S14RNHlWzX0D2dbAare1FMQjQ8Q0Fat4rj0lD/X2HfIbip9Zi49qTkco
+kzeGj/oamAA/bBH7xHX8q7QmlsYTuhykL4aSF16ufEvIvuCqP8eIBYFrkiIJGBS
aUSr1gIcFq7cba0x7zBHofyo9z3Fwg/k+rC119NAAz2CU5cJPHy6qpR0FwU0WmLK
oYLAQVOtuPAoL71De8xwGfqBMILzYlUse9FoP6Wix8aWHxfiv0ss7fgV6XVZOdz/
9prbW6K/Xu6r/jGen0+K3UH7n94mwRunAgMBAAGjTzBNMA4GA1UdDwEB/wQEAwIC
pDAdBgNVHSUEFjAUBggrBgEFBQcDAQYIKwYBBQUHAwIwDwYDVR0TAQH/BAUwAwEB
/zALBgNVHREEBDACggAwDQYJKoZIhvcNAQELBQADggEBAARxDSeHJRg0OcVuEQjI
EnMUKrMsk450Iu/7JPM7aXgVnJvCYA+75cpzbcJOVqJVMGR/dGUn+tnUZbfmoedF
tKWJDHS3dyk6Ko/gcnPH4bGaz7IsnuS10fsD159DRK30FxwWHjyCK3/5ubuXECTR
9KJ/1IzaO6WtqHw/pPEHB6WMFOWhRR060jBkOv750P64XSoJoQSZRbmTgU0uqm+H
jd3lbSjE0LMbisEuj6SVchlqe1IEyxFfZ3HfnvAxzuTQZ1QkN243GDC1FTKUaQt3
nJ65FGBrW9Wkx7NvpadCY/w0+uvSGCVn8ukPoNTAhL68S8xnqpUOgN+gm8zqjch6
/NI=
-----END CERTIFICATE-----
Bag Attributes
    friendlyName: localhost
    2.16.840.1.113894.746875.1.1: <Unsupported tag 6>
subject=C = Unknown, ST = Unknown, L = Unknown, O = Unknown, OU = Unknown, CN = Unknown

issuer=C = AU, ST = Some-State, O = Internet Widgits Pty Ltd

-----BEGIN CERTIFICATE-----
MIIE1zCCBEACFFFbBitYlpLZhXmZBKO4nOUvyr10MA0GCSqGSIb3DQEBCwUAMEUx
CzAJBgNVBAYTAkFVMRMwEQYDVQQIEwpTb21lLVN0YXRlMSEwHwYDVQQKExhJbnRl
cm5ldCBXaWRnaXRzIFB0eSBMdGQwHhcNMTkwNTE1MTkwMjM1WhcNMjAwNTE0MTkw
MjM1WjBsMRAwDgYDVQQGEwdVbmtub3duMRAwDgYDVQQIEwdVbmtub3duMRAwDgYD
VQQHEwdVbmtub3duMRAwDgYDVQQKEwdVbmtub3duMRAwDgYDVQQLEwdVbmtub3du
MRAwDgYDVQQDEwdVbmtub3duMIIDQjCCAjUGByqGSM44BAEwggIoAoIBAQCPeTXZ
uarpv6vtiHrPSVG28y7FnjuvNxjo6sSWHz79NgbnQ1GpxBgzObgJ58KuHFObp0db
hdARrbi0eYd1SYRpXKwOjxSzNggooi/6JxEKPWKpk0U0CaD+aWxGWPhL3SCBnDcJ
oBBXsZWtzQAjPbpUhLYpH51kjviDRIZ3l5zsBLQ0pqwudemYXeI9sCkvwRGMn/qd
gYHnM423krcw17njSVkvaAmYchU5Feo9a4tGU8YzRY+AOzKkwuDycpAlbk4/ijsI
OKHEUOThjBopo33fXqFD3ktm/wSQPtXPFiPhWNSHxgjpfyEc2B3KI8tuOAdl+CLj
Qr5ITAV2OTlgHNZnAh0AuvaWpoV499/e5/pnyXfHhe8ysjO65YDAvNVpXQKCAQAW
plxYIEhQcE51AqOXVwQNNNo6NHjBVNTkpcAtJC7gT5bmHkvQkEq9rI837rHgnzGC
0jyQQ8tkL4gAQWDt+coJsyB2p5wypifyRz6Rh5uixOdEvSCBVEy1W4AsNo0fqD7U
ielOD6BojjJCilx4xHjGjQUntxyaOrsLC+EsRGiWOefTznTbEBplqiuH9kxoJts+
xy9LVZmDS7TtsC98kOmkltOlXVNb6/xF1PYZ9j897buHOSXC8iTgdzEpbaiH7B5H
SPh++1/et1SEMWsiMt7lU92vAhErDR8C2jCXMiT+J67ai51LKSLZuovjntnhA6Y8
UoELxoi34u1DFuHvF9veA4IBBQACggEAQ2+qNgD+nRPMCHp/GRNDrhtcGBzKqmhR
biICW5t3SJ/r2RUdtIWmE4hYgptHUO5H1IxyImDZWjcvolxTHbQxbyMwrAgO4btT
IF43LOKL2kqJofjDorMb2YzqPeAjyQmTNSIDIy7nBVOmIl6gIrQhDXaAGtddDytp
DLjrjdMuCUF8evdqEoNVHmVuTZXWqB4qMlnpt9T1wlCXL5U6BsAYM5xr1srPvtw2
UGlEFoAchOp2EAs7KmwkGusBgZCxSKNPxVSFkTPhJYhaBz1tjxsELFSBZ9YZ9OtB
yid9PHVgMZ4wuyM4S1HYs7rHlY4ndkv3roYxAxuRxtL55cWZhybmJDANBgkqhkiG
9w0BAQsFAAOBgQBeoWGNw0JQN/SR4pccwhqY3CamAiFPMaz2kT7vmcFSfqJP0k/B
v+fqh9et2sFy69RmVkPrZSdLzArYPxQfzy1MENSLYrbiXXqpXzI4u8/k9/F4xYQ7
oWAjJrk5muE/7I7kZpF9s42O70pH+FBz+dCfJuReevLD3F9vYGHEKGJc8A==
-----END CERTIFICATE-----
```

Using UnifiedAutomation.UaBase.dll | Runtime Version v4.0.30319 | Version 2.0.0.0

AlinMoldovean, 2019-05-31:
@Never7Dice ,

You can use some `openssl `commands to convert certificates to/from different formats.
Also this repository has a `CertificateFactory.ExportCertificateAsPEM()` method.
A usage example can be found in "_SampleApplications\Samples\GDS\Client\Controls\ApplicationCertificateControl.cs"_

Never7Dice, 2019-06-03:
@AlinMoldovean, I ended up using InternalCertificate property of ICertificate. The solution below was sufficient for my case.

```
public static string MainCode()
{
    trustedCertificatesList = CertificateStore.Select(t => t.InternalCertificate).ToList();
    File.WriteAllText("truststore.cer.pem", CryptoHelper.ExportTrustStore(trustedCertificatesList));
}

public static string ExportTrustStore(List<X509Certificate2> trustedCertificates)
{
    var builder = new StringBuilder();
    foreach (var certificate in trustedCertificates)
    {
        builder.AppendLine("Bag Attributes");
        builder.AppendLine($"    friendlyName: {certificate.FriendlyName}");
        builder.AppendLine($"subject={certificate.Subject}");
        builder.AppendLine();
        builder.AppendLine($"issuer={certificate.Issuer}");
        builder.AppendLine();
        builder.AppendLine(ExportPublicKey(certificate));
    }
    return builder.ToString();
}

public static string ExportPublicKey(X509Certificate2 cert)
{
    var builder = new StringBuilder();

    builder.AppendLine("-----BEGIN CERTIFICATE-----");
    builder.AppendLine(Convert.ToBase64String(cert.Export(X509ContentType.Cert), Base64FormattingOptions.InsertLineBreaks));
    builder.AppendLine("-----END CERTIFICATE-----");

    return builder.ToString();
}
```

Also for some reason I couldn't find this sample in the lib I use CertificateFactory.ExportCertificateAsPEM().

closed 2019-06-03 by Never7Dice