OPCFoundation/UA-.NETStandard #735: Monitored Item - exception-based model [Question]

rudoq32, 2019-06-05:
In OP UA Part 4 [p. 60] 5.12.1.2 Sampling interval they say that _The Server may support data that is collected based on a sampling model or generated based on an **exception-based model**. The fastest supported sampling interval may be equal to 0, which indicates that the data item is exception-based rather than being sampled at some period. An exception-based model means that the underlying system does not require sampling and reports data changes._

That requires something like this (bellow), right?
`serverConfig->publishingIntervalLimits.min = 50.0;`
`serverConfig->publishingIntervalLimits.max = 3600.0 * 1000.0;`

`serverConfig->samplingIntervalLimits.min = 0.0;	// zero means event/exception based based`
`serverConfig->samplingIntervalLimits.max = 24.0 * 3600.0 * 1000.0;`

But once I tried to create monitored item with 
`UA_MonitoringMode monitoringMode = UA_MONITORINGMODE_REPORTING;`
`UA_TimestampsToReturn timestampsToReturn = UA_TIMESTAMPSTORETURN_BOTH;`
`UA_Double samplingInterval = 0.0;`
`UA_Int32 queueSize = 0;`
`// Configuration for subscription`
`UA_Double requestedPublishingInterval = 50.0;`
I am getting _BadInternalError_. I used UA_ENABLE_SUBSCRIPTIONS_EVENTS, UA_ENABLE_AMALGAMATION, UA_ARCHITECTURE_FREERTOSLWIP (among others).

Does it mean that my monitored item has to be Event based type?

Thanks Rudo

AlinMoldovean, 2019-06-21:
In the server application you can set the value of some variable nodes when a data change is reported by the underlying system, instead of using a pooling mechanism for sampling the data. This is the case of exception-based model.

rudoq32, 2019-06-27:
Hi @AlinMoldovean,
In my implementation (embedded system) I am using external interrupt to set node variable value. 

But if I understand monitoring item concept correctly the sampling interval of monitored item says how often server checks for the change. Does it mean to achieve "exception based model" I need to "raise an server event" to let server know about the change?

AlinMoldovean, 2019-07-22:
@rudoq32 ,

All you need to do in the server application is to call `ClearChangeMasks()` on the variable node after setting the new value.
The server library will has all the internal implementation of the logic that triggers the required notifications to client monitoring the value of that variable node.
An example can be found in `ReferenceNodeManager.DoSimulation()`

```
private void DoSimulation(object state)
        {
            try
            {
                lock (Lock)
                {
                    foreach (BaseDataVariableState variable in m_dynamicNodes)
                    {
                        variable.Value = GetNewValue(variable);
                        variable.Timestamp = DateTime.UtcNow;
                        variable.ClearChangeMasks(SystemContext, false);
                    }
                }
```

closed 2019-09-26 by AlinMoldovean