OPCFoundation/UA-.NETStandard #736: Use Custom type in CreateAddressSpace Programmatically

myasir009, 2019-06-06:
I am new to the OPC UA world. Can anybody guide my how to create a custom complex type in c# using OPCFoundation/UA-.NETStandard.In my CreateAddressSpace method I want to create a node 'ABC' with custom datatype like Person and Person will have some attributes or properties like name, age, id,Status with string, int and Enum data types. I want to done all this by code without using any code generating tool.

Also How can i create an use a Custom Enum in address space using code without any tool. I want to create Enum i.e Status enum with Single,window, married, divorce etc.

btmmf, 2019-06-07:
I had the same problem. Based on my research, I have the impression that it requires a custom Enumeration DataType in order to specify the names of the individual enumeration values.

As I also struggled to find out how to do this, I found the following workaround:
The following line creates a so called "multi state discrete item" variable.
To me this sounded exactly like the definition of what an enum is.
`CreateMultiStateDiscreteItemVariable(root, "Status", "Status", "Single", "Widow", "Married", "Divorced");`

Guess where the string array `["Single", "Widow", "Married", "Divorced"]` ends up being?
Yeah, right, it is stored in the `EnumStrings` property of the `Status` variable. Just like you would expect it from an actual enum variable. (expand the "Status" variable in your OPC UA client to see it)

It remains unclear to me why anybody would want to use a real enum instead, just to go through the hassle of creating a custom type, etc.

AlinMoldovean, 2019-06-10:
@myasir009 ,

Creating custom data types with this stack without using the support of Model Compiler is quite difficult and there is no sample code provided for this.

closed 2019-09-26 by AlinMoldovean