OPCFoundation/UA-.NETStandard #737: The AlarmCondition Client/Server pair in the UA Quickstarts Applications doesn't do anything (especially not, what the documentation shows)

btmmf, 2019-06-07:
Hi

I opened the UA Quickstart Applications solution twice in Visual Studio, to start the AlarmCondition Client and Server respectively.

The quickstart came to a grinding halt, as nothing was actually happening after establishing the connection except the applications being started.

[image]

According to the [documentation](http://opcfoundation.github.io/UA-.NETStandard/help/acclientoverview.htm), I was expecting some kind of output like this:

[image]

Is my expectation wrong? What is supposed to happen in this example? Am I supposed to do anything additional to connecting the client with the server?
Is the documentation outdated?
Is there a bug?

thanks for your help

btmmf, 2019-06-12:
Can anybody help me with this issue in any way?

AlinMoldovean, 2019-06-21:
@btmmf ,

The provided sample projects are just some basic examples of implementation for different features. They are nut fully functional and they need to be further extended.
You need to debug inside the code and find the reason why event notification are not triggered for alarms.

btmmf, 2019-06-21:
@AlinMoldovean 

See, I've tried that. But without a clear idea about what to expect, it's not easy to do.
Hence my question about the documentation and what the example is atually supposed to be doing.

Again, the documentation shows screenshots of some behaviour, which isn't actually happening, but obviously happened to create the screenshot. Can you help me with my actual problem instead of stating the obvious, namely that I have the option to help myself and that the example isn't working?

PaulHunkar, 2019-06-24:
So for alarms & condition the client provides options to specify a filter for what condition are desired (conditions drop down).  I have not looked at this example for several years, but if the filter is set in a manner that no events meet the filter - nothing will be displayed.  You might want to try  some different filter options and see if anything changes.  Also Alarms have a state, so in some case you might need a "Refresh" to get the current list of alarms.  The first two screen you posted show that the server is reporting a connection from the alarm client and that you have created a subscription (the alarm client as I remember only creates a subscription for events) - so the only issue could be the filter you have on the Subscription.

btmmf, 2019-06-25:
Here's how the Condition Type is configured:
[image]

And here's the Minimum Severity:
[image]

I'm pretty sure this configuration should show everything that comes up. But there's still **nothing**. Hiting **Conditions -> Refresh** does not help. What should I change?

The Area Filter shows no options to select. Selecting **Server** does not improve the situation.
[image]

This is different from the screenshot in the [documentation](http://opcfoundation.github.io/UA-.NETStandard/help/acclientoverview.htm):
[image]

@PaulHunkar @AlinMoldovean can you reproduce the issue?

Adasq01, 2019-08-21:
Hi @btmmf 

I had the same issue as you described here. It looks like there is a problem with generating server areas.

I did some debugging and find out that null reference exception is throwing up during invoke of `CreateDialog()` and `CreateAlarm()` methods inside `SourceState.cs` file when the `node.LocalTime.Value` is being set. I just commented it and everything starts to work as in documentation.

    // initialize event information.node
            node.EventType.Value = node.TypeDefinitionId;
            node.SourceNode.Value = this.NodeId;
            node.SourceName.Value = this.SymbolicName;
            node.ConditionName.Value = node.SymbolicName;
            node.Time.Value = DateTime.UtcNow; 
            node.ReceiveTime.Value = node.Time.Value;
            // node.LocalTime null reference exception
            // node.LocalTime.Value = Utils.GetTimeZoneInfo();
            node.BranchId.Value = branchId;`
`

Maybe it will also help you before the real solution will came.

theonlyrule, 2020-03-31:
Thanks. You can also change the code like this:
node.LocalTime = new PropertyState<TimeZoneDataType>(node.EnabledState);
node.LocalTime.Value = Utils.GetTimeZoneInfo();

commit 7f9689f2 (2020-03-31): Fixes #737 - null reference in AlarmCondition server. LocalTime property became optional.
commit f4853f25 (2020-03-31): Merge pull request #935 from AlinMoldovean/master

Fixes #737 - null reference in AlarmCondition server.
closed 2020-03-31 by AlinMoldovean