OPCFoundation/UA-.NETStandard #741: Not able to connect to UA Sever in None Mode and user identity token 'username password'

chandru9050, 2019-06-11:
I am unable to establish a UA Session using the dot net core OPC UA Client, I am connecting to None mode with user identity token as user name password.   I am using a self signed certificate and the server certificate is placed in client's trusted folder.   Error I am getting is BadCertificateUntrusted. in Certificate validator.cs  in below line of code,

  // check if certificate is trusted.
            if (trustedCertificate == null && !isIssuerTrusted)
            {
                if (m_applicationCertificate == null || !Utils.IsEqual(m_applicationCertificate.RawData, certificate.RawData))
                {
                    throw ServiceResultException.Create(
                        StatusCodes.BadCertificateUntrusted,
                        "Certificate is not trusted.\r\nSubjectName: {0}\r\nIssuerName: {1}",
                        certificate.SubjectName.Name,
                        certificate.IssuerName.Name);
                }
            }

AlinMoldovean, 2019-06-11:
@chandru9050 ,

The server certificate seems to be issued by a CA.
In this case the CA certificate should be in the trust list.

chandru9050, 2019-06-11:
@AlinMoldovean ,

I confirmed it, The server is using a self signed cert.

chandru9050, 2019-06-11:
@AlinMoldovean , Also this was working before with the older version of OPC UA SDK, only after upgrading to new SDK the problem is coming

AlinMoldovean, 2019-06-20:
@chandru9050 ,

Can you check the reason why _trustedCertificate_ is null? i.e. _GetTrustedCertificate()_ returns null at

https://github.com/OPCFoundation/UA-.NETStandard/blob/5c0d86873b5903206d7650c00c9646ee15770622/Stack/Opc.Ua.Core/Security/Certificates/CertificateValidator.cs#L698

closed 2019-09-26 by AlinMoldovean