OPCFoundation/UA-.NETStandard #744: No expiration/revocation check for validated certificate

ccasteyde, 2019-06-17:
When a certificate was validated once, it is added to the m_validatedCertificates cache forever (well unless the UA configuration file is modified), and it is not checked against expiration or revocation anymore. This is not good.
I consider adding a fix to set up a validity time window for the entries in the cache.

ccasteyde, 2019-06-17:
Fix is simple in CertificateValidator.cs:
Change m_validatedCertificates declaration to:
        private Dictionary<string,Tuple<X509Certificate2, DateTime>> m_validatedCertificates;
        //private Dictionary<string, X509Certificate2> m_validatedCertificates;
Then add the certificate in cache with expiration date:
                    //m_validatedCertificates[certificate.Thumbprint] = certificate;
                    m_validatedCertificates[certificate.Thumbprint] = new Tuple<X509Certificate2, DateTime>(certificate, DateTime.UtcNow.AddMinutes(10));

and then use the date to check expiracy when validating the certificate:
InternalValidate* 
                Tuple<X509Certificate2, DateTime> certificate2;
                //X509Certificate2 certificate2 = null;
                if (m_validatedCertificates.TryGetValue(certificate.Thumbprint, out certificate2))
                {
                    if (DateTime.UtcNow > certificate2.Item2)
                        m_validatedCertificates.Remove(certificate.Thumbprint);
                    else
                    if (Utils.IsEqual(certificate2.Item1.RawData, certificate.RawData))
                    //if (Utils.IsEqual(certificate2.RawData, certificate.RawData))
                    {
                        return;
                    }
                }

AlinMoldovean, 2019-06-20:
@ccasteyde ,

You can also clear the list of validated certificates at application level if needed with a call like this:

```
configuration.CertificateValidator.Update(configuration.SecurityConfiguration).Wait();
```

closed 2019-09-26 by AlinMoldovean