OPCFoundation/UA-.NETStandard #745: clear text username and password

Ramin2c, 2019-06-17:
Hi
Not sure if this is the right place to ask this question but I wonder if from the standard perspective it is OK to send clear text, non encrypted username and password from the client to the server over an OPC UA standard? I am new to OPC UA and if I have understood correctly there are things that are doable using OPC UA but might be against the standard, and the question I have asked is my fear that by sending clear text username and password our eventual system might not become OPC UA fully compliant.

Thanks
Ramin

btmmf, 2019-06-17:
You can do that. It is explicitly mentioned in the standard. It is then up to you to get the security from another part of the system, e.g. by securing the transport layer.

Check the following image from [here](http://www.ascolab.com/en/technology-unified-architecture/technology-protocols.html).
[image]

PaulHunkar, 2019-06-17:
If a product is to be certified, the password must be encrypted.  If using HTTPS for the transport than the transport will handle the encryption, if using TCP (Native UA Binary) than the application must perform the encryption.  If an application does not support security (only the Nano Device Profile which is for sensor type devices) then it does not typically support username/password either.  For additional detail on profile and what is required see https://apps.opcfoundation.org/profilereporting/

AlinMoldovean, 2019-06-19:
@Ramin2c ,

A compliant server should expose a configuration that requires the token types that contain sensitive data like user/password to be encrypted when exchanged between client and server.

closed 2019-09-19 by AlinMoldovean