OPCFoundation/UA-.NETStandard #750: Invalid default ValueRank for some variable types

saurla, 2019-06-25:
All variable types which ValueRank should be -2 (Any) have ValueRank 0 (OneOrMoreDimensions) by default. 

This is caused by a bug in the ModelCompiler which is now fixed (see issue https://github.com/OPCFoundation/UA-ModelCompiler/issues/27 and commit https://github.com/OPCFoundation/UA-ModelCompiler/commit/5f2afa3f388efee5596ffa874b046a18ade42246 to the files **BaseVariableState** and **BaseVariableTypeState**).

So this issue can be fixed by updating to the newest version of the UA Nodeset.

AlinMoldovean, 2019-07-22:
This issue should be fixed when updating the latest UA Nodesets and generated code from _UA-Nodeset_  repository.

AlinMoldovean, 2019-09-26:
Waiting for #791

AlinMoldovean, 2019-10-03:
Nodeset files updated in #791

closed 2019-10-03 by AlinMoldovean