OPCFoundation/UA-.NETStandard #753: `TypeDefinitionId` NULL when `session.ReadNode`

tomsoftware, 2019-07-10:
Why does the `session.ReadNode(nodeId).TypeDefinitionId` return NULL but `session.Browse` returns a value for `nextRd.TypeDefinition`.

Example from the \UA-.NETStandard\SampleApplications\Samples\NetCoreConsoleClient
Using `Objects / Boilers / Boiler #1` :: "ns=4;i=1241" as test node:

```
Console.WriteLine("4 - Browse the OPC UA server namespace.");
exitCode = ExitCode.ErrorBrowseNamespace;
ReferenceDescriptionCollection references;
Byte[] continuationPoint;

references = session.FetchReferences(ObjectIds.ObjectsFolder);

var boiler1NodeId = new NodeId(1241, 4);
var nodeInfo = session.ReadNode(boiler1NodeId);
Console.WriteLine("   ReadNode: {0} -> {1}", nodeInfo.BrowseName, nodeInfo.TypeDefinitionId);

session.Browse(
	null,
	null,
	ObjectIds.ObjectsFolder,
	0u,
	BrowseDirection.Forward,
	ReferenceTypeIds.HierarchicalReferences,
	true,
	(uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method,
	out continuationPoint,
	out references);

			 
foreach (var rd in references)
{
	ReferenceDescriptionCollection nextRefs;
	byte[] nextCp;
	session.Browse(
		null,
		null,
		ExpandedNodeId.ToNodeId(rd.NodeId, session.NamespaceUris),
		0u,
		BrowseDirection.Forward,
		ReferenceTypeIds.HierarchicalReferences,
		true,
		(uint)NodeClass.Variable | (uint)NodeClass.Object | (uint)NodeClass.Method,
		out nextCp,
		out nextRefs);

	foreach (var nextRd in nextRefs)
	{
		if (boiler1NodeId == nextRd.NodeId) {
			Console.WriteLine("   Browse {0} -> {1}", nextRd.BrowseName, nextRd.TypeDefinition);
		}
	}
}

```

Output:

[image]

AlinMoldovean, 2019-07-11:
@tomsoftware ,

The _session.ReadNode()_ method reads all the available attributes of the server node.
But _TypeDefinitionId_ information cannot be retrieved using a _Read_ service call because is not a standard OPC UA attribute. 
It can only be retrieved with a _Browse_ call and it only applies for _Object_ and _Variable_ node classes.

tomsoftware, 2019-07-11:
OK, I was confused, I thought `Attributes.DataTypeDefinition` applies to the `TypeDefinitionId` property. So `session.ReadNode` only reads the attributes of a Node and to get the TypeDefinition of a node, I need to call "session.FetchReferences" cause "TypeDefinition" is a reference (=HasTypeDefinition) not a attribute. 
Right?

But what for the `Node.TypeDefinitionId` property is, if it's ?always? empty...

Sorry for my misunderstanding and thanks for your help / work :-)

AlinMoldovean, 2019-07-22:
@tomsoftware ,

I think the `Node.TypeDefinitionId` property will be set in case the references from the attached `ReferenceTable` contains the _HasTypeDefinition_ references.
That can be set in the same way as NodeCache.FetchNode() method.
```
        public Node FetchNode(ExpandedNodeId nodeId)
        {
            NodeId localId = ExpandedNodeId.ToNodeId(nodeId, m_session.NamespaceUris);

            if (localId == null)
            {
                return null;
            }

            // fetch node from server.
            Node source = m_session.ReadNode(localId);

            try
            {
                // fetch references from server.
                ReferenceDescriptionCollection references = m_session.FetchReferences(localId);

                foreach (ReferenceDescription reference in references)
                {
                    // create a placeholder for the node if it does not already exist.
                    if (!m_nodes.Exists(reference.NodeId))
                    {
                        // transform absolute identifiers.
                        if (reference.NodeId != null && reference.NodeId.IsAbsolute)
                        {
                            reference.NodeId = ExpandedNodeId.ToNodeId(reference.NodeId, NamespaceUris);
                        }

                        Node target = new Node(reference);
                        m_nodes.Attach(target);
                    }

                    // add the reference.
                    source.ReferenceTable.Add(reference.ReferenceTypeId, !reference.IsForward, reference.NodeId);

```

tomsoftware, 2019-08-03:
Thank you - I will try this

closed 2019-08-03 by tomsoftware