OPCFoundation/UA-.NETStandard #754: OnCall method is not being invoked(Error: BadArgumentMissing)

NakWarsi, 2019-07-12:
I have written Information model in XML file Where I have defined one Method State,
That method State is being registered with `GenericMethodCalledEventHandler` but if I am calling this method by a client I am getting error "BadArgumentMissing"

XML of Method State with the arguments:-
`
  <UAMethod SymbolicName="Method_State" NodeId="ns=1;i=50061" BrowseName="1:MethodState">
    <DisplayName>Method State</DisplayName>
    <References>
      <Reference ReferenceType="HasProperty">ns=1;i=50062</Reference>
      <Reference ReferenceType="HasComponent" IsForward="false">ns=1;i=5006</Reference>
    </References>
  </UAMethod>
  <UAVariable DataType="Argument" ParentNodeId="ns=1;i=50061" ValueRank="1" NodeId="ns=1;i=50062" ArrayDimensions="8" BrowseName="InputArguments">
    <DisplayName>InputArguments</DisplayName>
    <References>
      <Reference ReferenceType="HasProperty" IsForward="false">ns=1;i=50061</Reference>
      <Reference ReferenceType="HasTypeDefinition">i=68</Reference>
    </References>
    <Value>
      <uax:ListOfExtensionObject>
        <uax:ExtensionObject>
          <uax:TypeId>
            <uax:Identifier>i=297</uax:Identifier>
          </uax:TypeId>
          <uax:Body>
            <uax:Argument>
              <uax:Name>IntegerValue</uax:Name>
              <uax:DataType>
                <uax:Identifier>i=11</uax:Identifier>
              </uax:DataType>
              <uax:ValueRank>-1</uax:ValueRank>
              <uax:ArrayDimensions/>
              <uax:Description/>
            </uax:Argument>
          </uax:Body>
        </uax:ExtensionObject>
        <uax:ExtensionObject>
          <uax:TypeId>
            <uax:Identifier>i=297</uax:Identifier>
          </uax:TypeId>
          <uax:Body>
            <uax:Argument>
              <uax:Name>BuildYear</uax:Name>
              <uax:DataType>
                <uax:Identifier>i=11</uax:Identifier>
              </uax:DataType>
              <uax:ValueRank>-1</uax:ValueRank>
              <uax:ArrayDimensions/>
              <uax:Description/>
            </uax:Argument>
          </uax:Body>
        </uax:ExtensionObject>
        <uax:ExtensionObject>
          <uax:TypeId>
            <uax:Identifier>i=297</uax:Identifier>
          </uax:TypeId>
          <uax:Body>
            <uax:Argument>
              <uax:Name>LocationMachine</uax:Name>
              <uax:DataType>
                <uax:Identifier>i=11</uax:Identifier>
              </uax:DataType>
              <uax:ValueRank>-1</uax:ValueRank>
              <uax:ArrayDimensions/>
              <uax:Description/>
            </uax:Argument>
          </uax:Body>
        </uax:ExtensionObject>
        <uax:ExtensionObject>
          <uax:TypeId>
            <uax:Identifier>i=297</uax:Identifier>
          </uax:TypeId>
          <uax:Body>
            <uax:Argument>
              <uax:Name>LocationPlant</uax:Name>
              <uax:DataType>
                <uax:Identifier>i=11</uax:Identifier>
              </uax:DataType>
              <uax:ValueRank>-1</uax:ValueRank>
              <uax:ArrayDimensions/>
              <uax:Description/>
            </uax:Argument>
          </uax:Body>
        </uax:ExtensionObject>
        <uax:ExtensionObject>
          <uax:TypeId>
            <uax:Identifier>i=297</uax:Identifier>
          </uax:TypeId>
          <uax:Body>
            <uax:Argument>
              <uax:Name>Manufacturer</uax:Name>
              <uax:DataType>
                <uax:Identifier>i=11</uax:Identifier>
              </uax:DataType>
              <uax:ValueRank>-1</uax:ValueRank>
              <uax:ArrayDimensions/>
              <uax:Description/>
            </uax:Argument>
          </uax:Body>
        </uax:ExtensionObject>
        <uax:ExtensionObject>
          <uax:TypeId>
            <uax:Identifier>i=297</uax:Identifier>
          </uax:TypeId>
          <uax:Body>
            <uax:Argument>
              <uax:Name>NameCatalog</uax:Name>
              <uax:DataType>
                <uax:Identifier>i=11</uax:Identifier>
              </uax:DataType>
              <uax:ValueRank>-1</uax:ValueRank>
              <uax:ArrayDimensions/>
              <uax:Description/>
            </uax:Argument>
          </uax:Body>
        </uax:ExtensionObject>
        <uax:ExtensionObject>
          <uax:TypeId>
            <uax:Identifier>i=297</uax:Identifier>
          </uax:TypeId>
          <uax:Body>
            <uax:Argument>
              <uax:Name>NameCustom</uax:Name>
              <uax:DataType>
                <uax:Identifier>i=11</uax:Identifier>
              </uax:DataType>
              <uax:ValueRank>-1</uax:ValueRank>
              <uax:ArrayDimensions/>
              <uax:Description/>
            </uax:Argument>
          </uax:Body>
        </uax:ExtensionObject>
        <uax:ExtensionObject>
          <uax:TypeId>
            <uax:Identifier>i=297</uax:Identifier>
          </uax:TypeId>
          <uax:Body>
            <uax:Argument>
              <uax:Name>SerialNumber</uax:Name>
              <uax:DataType>
                <uax:Identifier>i=11</uax:Identifier>
              </uax:DataType>
              <uax:ValueRank>-1</uax:ValueRank>
              <uax:ArrayDimensions/>
              <uax:Description/>
            </uax:Argument>
          </uax:Body>
        </uax:ExtensionObject>
      </uax:ListOfExtensionObject>
    </Value>
  </UAVariable>`

C# code to handle this:-

`protected override NodeState AddBehaviourToPredefinedNode(ISystemContext context, NodeState predefinedNode)
        {

            BaseObjectState passiveNode = predefinedNode as BaseObjectState;

            if (passiveNode == null)
            {

                if (predefinedNode is MethodState passiveMethod)
                {
                    passiveMethod.OnCallMethod = new GenericMethodCalledEventHandler(OnStartCall);
                    return passiveMethod;
                }
            }

            return predefinedNode;
        }

        private ServiceResult OnStartCall(
            ISystemContext context,
            MethodState method,
            IList<object> inputArguments,
            IList<object> outputArguments)
        {
            return ServiceResult.Good;
        }`

Error:-
[image]

Please Help,

AlinMoldovean, 2019-07-22:
@NakWarsi ,

You need to debug inside the StandardServer.Call() method and find the reason of the returned bad status code:

```
  public override ResponseHeader Call(
            RequestHeader                  requestHeader,
            CallMethodRequestCollection    methodsToCall,
            out CallMethodResultCollection results,
            out DiagnosticInfoCollection   diagnosticInfos)
        {
            OperationContext context = ValidateRequest(requestHeader, RequestType.Call);
                
            try
            {
                if (methodsToCall == null || methodsToCall.Count == 0)
                {
                    throw new ServiceResultException(StatusCodes.BadNothingToDo);
                }         

                m_serverInternal.NodeManager.Call(
                    context,
                    methodsToCall,
                    out results,
                    out diagnosticInfos);

                return CreateResponse(requestHeader, context.StringTable);
```

NakWarsi, 2019-10-01:
there was a mismatch in the argument of the server and Client.
so found the error and fixed,
I am closing this with a suggestion(for those who are might get the same error), one contract should be passed between Client and Server so things like that shouldn't happen.

Thank You @AlinMoldovean,

closed 2019-10-01 by NakWarsi