OPCFoundation/UA-.NETStandard #755: Historical access of a variable

Mark01010101, 2019-07-15:
Hi everyone,

I am working on a thesis relating to OPC UA. I have created a server for the thesis that supports data access using the .NET stack. I am using UA Expert as my client for testing.

I was trying to implement historizing of a variable for one of my variables as an example. I followed the Historical Access Server example and tried to use it to implement my own historizing functionality. The example however seems very complicated, especially for a person who is still fairly inexperienced in coding. The necessary information for creating a simple historical access functionality did not come clear.

Could someone provide a simplified code for implementing historical access of a variable? Thank you for the help.

MD-V, 2019-07-23:
Hello @Mark01010101,

I have created a little sample server (does not include aggregation, continuation points and such).

You can find it here [History Sample Server](https://gist.github.com/MD-V/4f7023d2c61cd94c1fa012260df6f43d)

You need the nuget package and C# 7.1 support enabled to run.

Mark01010101, 2019-07-23:
Hi @MD-V,

Thank you a lot for your help! This helped me a lot with implementing historizing to my server. I was able to implement a simple historical access functionality following your example code.

Only one thing I noticed:

I needed to change the AccessLevels in your example to HistoryRead to access the historizing of the variable with UAExpert
```
haState.AccessLevel = AccessLevels.CurrentRead | AccessLevels.HistoryRead;
haState.UserAccessLevel = AccessLevels.CurrentRead | AccessLevels.HistoryRead;
```

MD-V, 2019-07-26:
@Mark01010101
Strange, it worked for me without the change of access levels...

 Please close the issue if resolved

closed 2019-07-26 by Mark01010101

cannonkalra, 2020-05-07:
@MD-V I I'm not able to write any Value to the HAConfiguration Node

            configuration.StartOfArchive.Value = DateTime.Now;
            configuration.StartOfOnlineArchive.Value = DateTime.Now;

This doesn't work.. Need help..