OPCFoundation/UA-.NETStandard #758: What is the difference SourceTimeStamp and ServerTimeStamp in DataValue Object?

guzelcihad, 2019-07-16:
Hi,

I've writing an opc client. I've subscribe some tags in server and read these values.

`MonitoredItem item = new MonitoredItem();
IList<DataValue value = item.DequeValues()`;

The value object contains information about the value read from Server. And includes SourceTimestamp and ServerTimeStamp. What is the differences between them?

Is ServerTimestamp about, the time when the value sent from server to OPC Client?
Is SourceTimeStamp about, the time when the tag value read from server?

If yes, why these values are equal? Even miliseconds are equal.

Thanks

ejoseca, 2019-07-16:
1. The sourceTimestamp is used to reflect the timestamp that was applied to a Variable value by the data source.
2. The serverTimestamp is used to reflect the time that the Server received a Variable value or knew it to be accurate.

The serverTimestamp is set when the server to which you are connected gets the Variable value (by sampling or notification).
If the server is the one setting that value, serverTimestamp and sourceTimestamp will be the same.

guzelcihad, 2019-07-16:
Thanks for comment @ejoseca.

closed 2019-07-16 by guzelcihad