OPCFoundation/UA-.NETStandard #762: ***EXCEPTION*** BadNotConnected. RegisterServer failed for at: opc.tcp://localhost:4840/

commcad, 2019-07-18:
Hi Team

I've just taken on UA and planing an implementation of the .NETStandard in our company. My starting point is the Reference Server (.net 4.6). I am currently trying hard to set myself up with a stable environment and so far so good! However I am getting difficulties is registering the server with a discovery server, of which we don't need because we'll only have one server running LAN.

**My question;** how to disable the discovery lookup service on server start-up? I am getting a great deal of errors that hopefully could be disabled, if possible.

Unless there is something I am not doing incorrectly, although all seems to work flawlessly, apart from below. Any pointers would be appreciated. 

Thanks, (and great work btw). 

************************* Logging started at 07/19/2019 01:14:54
7/19/2019 01:14:54.245 Checking application instance certificate.
7/19/2019 01:14:54.251 Creating application instance certificate.
7/19/2019 01:14:54.254 Deleting application instance certificate.
7/19/2019 01:14:55.088 Created new certificate: 5C9A6E2E3490DF2DBE22BC1156770DA5815448F0
7/19/2019 01:14:55.108 Certificate created. Thumbprint=5C9A6E2E3490DF2DBE22BC1156770DA5815448F0
7/19/2019 01:14:55.377 The server is starting.
7/19/2019 01:14:55.734 Creating the Node Managers.
7/19/2019 01:14:56.407 Server: Session Monitor Thread Started.
7/19/2019 01:14:56.411 Server: Publish Subscriptions Thread Started.
7/19/2019 01:14:57.493 Channel 0 in Connecting state.
7/19/2019 01:14:58.544 ***EXCEPTION*** BadNotConnected
7/19/2019 01:14:58.597 ***EXCEPTION*** BadNotConnected 'Error establishing a connection: BadNotConnected'
7/19/2019 01:14:58.634 Channel 0: Close
7/19/2019 01:14:58.641 Channel 0 in Closed state.
7/19/2019 01:14:58.642 CLIENTCHANNEL SOCKET CLOSED: FFFFFFFF, ChannelId=0
7/19/2019 01:14:58.643 RegisterServer2 failed for at: opc.tcp://localhost:4840/. Exception=Error establishing a connection: BadNotConnected
7/19/2019 01:14:58.645 Channel 0 in Connecting state.
7/19/2019 01:14:59.650 ***EXCEPTION*** BadNotConnected
7/19/2019 01:14:59.680 ***EXCEPTION*** BadNotConnected 'Error establishing a connection: BadNotConnected'
7/19/2019 01:14:59.713 Channel 0: Close
7/19/2019 01:14:59.715 Channel 0 in Closed state.
7/19/2019 01:14:59.717 CLIENTCHANNEL SOCKET CLOSED: FFFFFFFF, ChannelId=0
7/19/2019 01:14:59.719 RegisterServer failed for at: opc.tcp://localhost:4840/. Exception=Error establishing a connection: BadNotConnected
7/19/2019 01:14:59.722 Register server failed. Trying again in 2000 ms
7/19/2019 01:15:01.800 Channel 0 in Connecting state.
7/19/2019 01:15:02.567 Channel 0 in Connecting state.
7/19/2019 01:15:02.809 ***EXCEPTION*** BadNotConnected
7/19/2019 01:15:02.820 ***EXCEPTION*** BadNotConnected 'Error establishing a connection: BadNotConnected'
7/19/2019 01:15:02.832 Channel 0: Close
7/19/2019 01:15:02.833 Channel 0 in Closed state.
7/19/2019 01:15:02.834 CLIENTCHANNEL SOCKET CLOSED: FFFFFFFF, ChannelId=0
7/19/2019 01:15:02.835 RegisterServer2 failed for at: opc.tcp://localhost:4840/. Exception=Error establishing a connection: BadNotConnected
7/19/2019 01:15:02.836 Channel 0 in Connecting state.
7/19/2019 01:15:03.576 ***EXCEPTION*** BadNotConnected
7/19/2019 01:15:03.616 ***EXCEPTION*** BadNotConnected 'Error establishing a connection: BadNotConnected'
7/19/2019 01:15:03.682 Channel 0: Close
7/19/2019 01:15:03.683 Channel 0 in Closed state.
7/19/2019 01:15:03.684 CLIENTCHANNEL SOCKET CLOSED: FFFFFFFF, ChannelId=0
7/19/2019 01:15:03.685 RegisterServer failed for at: opc.tcp://localhost:4840/. Exception=Error establishing a connection: BadNotConnected
7/19/2019 01:15:03.686 Channel 0 in Connecting state.
7/19/2019 01:15:03.844 ***EXCEPTION*** BadNotConnected
7/19/2019 01:15:03.863 ***EXCEPTION*** BadNotConnected 'Error establishing a connection: BadNotConnected'
7/19/2019 01:15:03.877 Channel 0: Close
7/19/2019 01:15:03.879 Channel 0 in Closed state.
7/19/2019 01:15:03.880 CLIENTCHANNEL SOCKET CLOSED: FFFFFFFF, ChannelId=0
7/19/2019 01:15:03.881 RegisterServer2 failed for at: opc.tcp://localhost:4840/. Exception=Error establishing a connection: BadNotConnected
7/19/2019 01:15:03.882 Register server failed. Trying again in 4000 ms
7/19/2019 01:15:04.688 ***EXCEPTION*** BadNotConnected
7/19/2019 01:15:04.711 ***EXCEPTION*** BadNotConnected 'Error establishing a connection: BadNotConnected'
7/19/2019 01:15:04.772 Channel 0: Close
7/19/2019 01:15:04.773 Channel 0 in Closed state.
7/19/2019 01:15:04.774 CLIENTCHANNEL SOCKET CLOSED: FFFFFFFF, ChannelId=0
7/19/2019 01:15:04.775 RegisterServer2 failed for at: opc.tcp://localhost:4840/. Exception=Error establishing a connection: BadNotConnected
7/19/2019 01:15:04.778 Server: Publish Subscriptions Thread Exited Normally.
7/19/2019 01:15:04.779 Server: Session Monitor Thread Exited Normally.

MD-V, 2019-07-19:
Hello @commcad,

set the value of ApplicationConfiguration.ServerConfiguration.MaxRegistrationInterval to 0 and it should never try to register itself.

commcad, 2019-07-19:
@MD-V
That worked flawlessly. Now having a perfectly stable Reference Server running without exceptions.
Thank you very much

closed 2019-07-19 by commcad

infowaddy, 2025-01-23:
I am curious: where does the OPC UA SDK get the Discovery Server Endpoint URL opc.tcp://localhost:4840? Is it hardcoded inside the SDK Standard Server library?