OPCFoundation/UA-.NETStandard #763: Xamarin.Android Ua Client Security Policy not supported

christhiel, 2019-07-20:
The attempt connecting the Xamarin.Android Client to a Server failed with the response, that the SecurityPolicy is not supported by the Client.
Probably, the reason is that `FieldInfo[] fields = typeof(SecurityPolicies).GetFields(BindingFlags.Public | BindingFlags.Static);`  in `Opc.Ua.SecurityPolicies.GetDisplayName(string policyUri)` returns an empty array.

Could it be, that `System.Reflection` doesn´t work correctly on an Android device?

AlinMoldovean, 2019-07-23:
@christhiel ,

That can be a Reflection issue on Xamarin.
You should try setting the list manually. e.g. add just `SecurityPolicies.None`

christhiel, 2019-07-23:
Thanks Alin! 
I´ve set the supported policies manually and so it works.

closed 2019-07-23 by christhiel