OPCFoundation/UA-.NETStandard #764: How to include type definitions in nodeset2.xml file

tschombe, 2019-07-22:
I am exporting my address space of the server by calling NodeStatCollection.SaveAsNodeSet2()
and creating a nodeset2.xml file.

But it only contains the node declarations and NOT the corresponding types
( i am also using custom types in my address space). Is there also a way to include the type definitions in the NodeSet2 file too?

AlinMoldovean, 2019-07-23:
Duplicate of #694

tschombe, 2019-07-23:
sorry for posting this duplicate, but since the other issue is already closed, i thought it would not get noticed. 

How can I include my own custom types? 
I would like to have one single file containing all node and corresponding type information

AlinMoldovean, 2019-07-23:
@tschombe ,

I already put a comment on the previous post [here](https://github.com/OPCFoundation/UA-.NETStandard/issues/694#issuecomment-513812145)

closed 2019-09-19 by AlinMoldovean